ZuluIDE/ZuluIDE-firmware #68: Latest Firmware unusable under Win98 SE | MTI R547 Motherboard | AMD K6 - III+ 450mhz

6ennadios, 2024-05-09:
Bear with me as I'm not a programmer, but the latest firmware passes POST, Win98 SE loads successfully, but hangs the system after the startup process, I'm assuming when the system tries the contents of the CD image. If I wait long enough the system loads the proper icon on the ISO image but assigns a random string of integers to the drive name. This behavior was not present with the original (shipped) firmware.

The logs from the last two boot attempts:

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.05.03-release May  3 2024 19:58:03
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D744BE3085062E6
[100ms] FPGA initialization succeeded
[117ms] SD card detected, FAT64 volume size: 60874 MB
[117ms] SD MID: 0x03, OID: 0x53 0x44
[117ms] SD Name: SN64G
[118ms] SD Date: 6/2022
[118ms] SD Serial: 0x79783944
[136ms] Loading first image CDRM_RRT2.iso
[140ms] Device is a CDROM drive
[140ms] Loading image CDRM_RRT2.iso
[150ms] Device 1 configuration:
[150ms] -- Max PIO mode: 3 (phy max 3)
[150ms] -- Max UDMA mode: 0 (phy max 0)
[151ms] -- Max blocksize: 4096 (phy max 4096)
[159ms] Initialization complete!
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x4B 0x74 0x2D 0xEF 0xFD 0xFF 0xFF 0x3A 0xE7 0x79 0x68 0x94 
[2000ms] FPGA license accepted with status 0x81
[115660ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[175877ms] -- WARNING: Unsupported ATAPI command ATAPI_CMD_SEEK10
[444710ms] WARNING: Detected supply voltage drop to 2937mV. Verify power supply is adequate.
```
```

[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.05.03-release May  3 2024 19:58:03
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D744BE3085062E6
[100ms] FPGA initialization succeeded
[117ms] SD card detected, FAT64 volume size: 60874 MB
[117ms] SD MID: 0x03, OID: 0x53 0x44
[117ms] SD Name: SN64G
[118ms] SD Date: 6/2022
[118ms] SD Serial: 0x79783944
[136ms] Loading first image CDRM_RRT2.iso
[140ms] Device is a CDROM drive
[140ms] Loading image CDRM_RRT2.iso
[150ms] Device 1 configuration:
[150ms] -- Max PIO mode: 3 (phy max 3)
[151ms] -- Max UDMA mode: 0 (phy max 0)
[151ms] -- Max blocksize: 4096 (phy max 4096)
[159ms] Initialization complete!
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x4B 0x74 0x2D 0xFF 0xFD 0xFF 0xFF 0x3A 0xE7 0x79 0x68 0x94 
[2000ms] FPGA license accepted with status 0x81
[38112ms] -- Set UDMA transfer mode 0
[542157ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[579955ms] -- WARNING: Unsupported ATAPI command ATAPI_CMD_SEEK10
[916399ms] WARNING: Detected supply voltage drop to 2958mV. Verify power supply is adequate.
```

aperezbios, 2024-05-09:
@6ennadios Thank you for this report. You wrote that "This behavior was not present with the original (shipped) firmware." but since we don't know what version that was, do you recall what that version of firmware was?

If you could try these previous releases of the firmware and let us know if the same problem manifests, that would be very helpful, as it would help us narrow down which change caused the regression in behavior:

* https://github.com/rabbitholecomputing/ZuluIDE-firmware/releases/download/v2024.04.23/ZuluIDE_RP2040_2024-04-23.uf2
* https://github.com/rabbitholecomputing/ZuluIDE-firmware/releases/download/v2024.04.08/ZuluIDE_RP2040_2024-04-08.uf2
* https://github.com/rabbitholecomputing/ZuluIDE-firmware/releases/download/v2024.04.03/ZuluIDE_RP2040_2024-04-03.uf2

morio, 2024-05-09:
@6ennadios you mentioned something about the primary IDE interface not having any issues with the ZuluIDE, am I reading that correctly? 

Yes and please find the working firmware, I can spin you some custom firmware to help narrow down the root cause if you are willing.

6ennadios, 2024-05-09:
Hello, the original firmware I received was 4.23
I moved my hardware setup back to IDE2 and reflashed the 4.23 firmware for expected behavior; the following logs are me restarting windows and cycling through all the cd images with the 4.23 bios.

@morio That is correct, the issues I'm having seem to be entirely related to running from IDE2.

I'm posting 2 dumps of 4.23 from IDE2, one normal one debug. And 2 more dumps from IDE2 when I reverted to 5.3

```
[12ms] Platform: ZuluIDE RP2040
[12ms] FW Version: 2024.04.23-release Apr 23 2024 18:42:24
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D744BE3085062E6
[101ms] FPGA initialization succeeded
[118ms] SD card detected, FAT64 volume size: 60874 MB
[118ms] SD MID: 0x03, OID: 0x53 0x44
[119ms] SD Name: SN64G
[119ms] SD Date: 6/2022
[119ms] SD Serial: 0x79783944
[138ms] Device is a CDROM drive
[138ms] Loading image CDRM_ALDO Shareware Disc.iso
[148ms] Device 1 configuration:
[148ms] -- Max PIO mode: 3 (phy max 3)
[149ms] -- Max UDMA mode: 0 (phy max 0)
[149ms] -- Max blocksize: 4096 (phy max 4096)
[157ms] Initialization complete!
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x4B 0x74 0x2D 0xFE 0xFF 0xFF 0xFF 0xA8 0xFC 0x79 0xA2 0xB2 
[2000ms] FPGA license accepted with status 0x80
[59366ms] -- Set UDMA transfer mode 0
[75478ms] -- Command handler failed for IDE_CMD_INIT_DEV_PARAMS
[86644ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[86654ms] -- WARNING: Unsupported ATAPI command ATAPI_CMD_SEEK10
[138892ms] Device ejecting media
[139171ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[144248ms] Device ejecting media
[144259ms] ---- Cue sheet CDRM_CIV3_DISK1.cue loaded with 1 tracks
[144503ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[170700ms] Device ejecting media
[170710ms] ---- Cue sheet CDRM_CIV3_Disk2.cue loaded with 1 tracks
[170954ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[178250ms] Device ejecting media
[178261ms] ---- Cue sheet CDRM_CIV3_Disk3.cue loaded with 1 tracks
[178503ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[181387ms] Device ejecting media
[181400ms] ---- Cue sheet CDRM_KODP.cue loaded with 1 tracks
[181622ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[189932ms] Device ejecting media
[190146ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[199727ms] Device ejecting media
[199974ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[214139ms] Device ejecting media
[216626ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[228934ms] Device ejecting media
[229124ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[239188ms] Device ejecting media
[239397ms] -- WARNING: Unsupported ATAPI command UNKNOWN_CMD
[287161ms] WARNING: Detected supply voltage drop to 2946mV. Verify power supply is adequate.
```
```
[12ms] Platform: ZuluIDE RP2040
[12ms] FW Version: 2024.04.23-release Apr 23 2024 18:42:24
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 1
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D744BE3085062E6
[14ms] DBG Loading FPGA bitstream
[101ms] FPGA initialization succeeded
[119ms] SD card detected, FAT64 volume size: 60874 MB
[119ms] SD MID: 0x03, OID: 0x53 0x44
[119ms] SD Name: SN64G
[119ms] SD Date: 6/2022
[... 2234 lines cut ...]
[257667ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x2F)
[257669ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x01 0x89 0x9A 0x00 0x00 0x01 0x00 0x00 0x00 
[257670ms] DBG -- Read 1 sectors starting at 100762
[257670ms] DBG ---- Read CD: 1 sectors starting at 100762, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 206360576
[257674ms] DBG -- ATAPI success
[257674ms] DBG -- Command complete
[257674ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x2F)
[257675ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x01 0x89 0x9B 0x00 0x00 0x06 0x00 0x00 0x00 
[257677ms] DBG -- Read 6 sectors starting at 100763
[257677ms] DBG ---- Read CD: 6 sectors starting at 100763, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 206362624
[257683ms] DBG -- ATAPI success
[257683ms] DBG -- Command complete
[258032ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x2F)
[258034ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x01 0x89 0xB8 0x00 0x00 0x02 0x00 0x00 0x00 
[258035ms] DBG -- Read 2 sectors starting at 100792
[258035ms] DBG ---- Read CD: 2 sectors starting at 100792, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 206422016
[258039ms] DBG -- ATAPI success
[258039ms] DBG -- Command complete
[258039ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x2F)
[258041ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x01 0x89 0xBA 0x00 0x00 0x01 0x00 0x00 0x00 
[258042ms] DBG -- Read 1 sectors starting at 100794
[258042ms] DBG ---- Read CD: 1 sectors starting at 100794, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 206426112
[258045ms] DBG -- ATAPI success
[258045ms] DBG -- Command complete
[260048ms] DBG

morio, 2024-05-10:
@6ennadios thanks for figuring out the working revision
Could you try this build? It will determine if the new fpga bit stream is the cause of this issue.
[ZuluIDE-4.23-with-new-fpga.zip](https://github.com/rabbitholecomputing/ZuluIDE-firmware/files/15278708/ZuluIDE-4.23-with-new-fpga.zip)

6ennadios, 2024-05-10:
Not working with slightly different behavior. Under 5.03 with debug off, Windows would identify the CD Rom but hang the system. With this revision windows fails to identify the CD Rom drive under both debug configurations.

```
[12ms] Platform: ZuluIDE RP2040
[12ms] FW Version: 2024.04.23-release May 10 2024 11:28:05
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D744BE3085062E6
[101ms] FPGA initialization succeeded
[118ms] SD card detected, FAT64 volume size: 60874 MB
[118ms] SD MID: 0x03, OID: 0x53 0x44
[119ms] SD Name: SN64G
[119ms] SD Date: 6/2022
[119ms] SD Serial: 0x79783944
[139ms] Device is a CDROM drive
[139ms] Loading image CDRM Black & White (USA) (Rerelease).bin
[151ms] ---- Cue sheet CDRM Black & White (USA) (Rerelease).cue loaded with 1 tracks
[160ms] Device 1 configuration:
[161ms] -- Max PIO mode: 3 (phy max 3)
[161ms] -- Max UDMA mode: 0 (phy max 0)
[161ms] -- Max blocksize: 4096 (phy max 4096)
[169ms] Initialization complete!
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x4B 0x74 0x2D 0xFF 0xFF 0xFF 0xDF 0x3A 0xE7 0x79 0x68 0x94 
[2000ms] FPGA license accepted with status 0x82
```
```
[12ms] Platform: ZuluIDE RP2040
[12ms] FW Version: 2024.04.23-release May 10 2024 11:28:05
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 1
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D744BE3085062E6
[14ms] DBG Loading FPGA bitstream
[101ms] FPGA initialization succeeded
[119ms] SD card detected, FAT64 volume size: 60874 MB
[119ms] SD MID: 0x03, OID: 0x53 0x44
[120ms] SD Name: SN64G
[120ms] SD Date: 6/2022
[120ms] SD Serial: 0x79783944
[139ms] Device is a CDROM drive
[139ms] Loading image CDRM Black & White (USA) (Rerelease).bin
[147ms] DBG Image file CDRM Black & White (USA) (Rerelease).bin is contiguous, sectors 19397888 to 21026050
[152ms] ---- Cue sheet CDRM Black & White (USA) (Rerelease).cue loaded with 1 tracks
[162ms] Device 1 configuration:
[162ms] -- Max PIO mode: 3 (phy max 3)
[162ms] -- Max UDMA mode: 0 (phy max 0)
[162ms] -- Max blocksize: 4096 (phy max 4096)
[171ms] DBG -- Operating as secondary drive
[171ms] Initialization complete!
[175ms] DBG IDE_EVENT_HWRST
[1176ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0xFF ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x4B 0x74 0x2D 0xEF 0xFD 0xFF 0xDF 0x3A 0xE7 0x79 0x68 0x94 
[2000ms] FPGA license accepted with status 0x81
[53228ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0xFF ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[53679ms] DBG IDE_EVENT_SWRST
[54680ms] DBG -- IDE regs: STATUS:0x00 CMD:0xEC DEV:0xA0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[56370ms] DBG IDE Command for DEV1: 0xA1 IDE_CMD_IDENTIFY_PACKET_DEVICE (device 0xB0, dev_ctrl 0x00, feature 0xAA, sector_count 0x01, lba 0xEB 0x14 0x01)
[56807ms] DBG -- Command complete
[56870ms] DBG IDE_EVENT_SWRST
[57871ms] DBG -- IDE regs: STATUS:0x00 CMD:0xEC DEV:0xA0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x01 LBAM:0x14 LBAH:0xEB
[59873ms] DBG -- IDE regs: STATUS:0x00 CMD:0xC6 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x00
[60874ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x00
[62876ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x00
[63877ms] DBG -- IDE regs: STATUS:0x00 CMD:0xC4 DEV:0x00 DEVCTRL:0x08 ERROR:0x01 FEATURE:0x00 LBAL:0x6F LBAM:0x8D LBAH:0x14
[64878ms] DBG -- IDE regs: STATUS:0x00 CMD:0xC4 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0xAF LBAM:0x00 LBAH:0x0A
[70884ms] DBG -- IDE regs: STATUS:0x00 CMD:0xC4 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0xCF LBAM:0x00 LBAH:0x00
[71885ms] DBG -- IDE regs: STATUS:0x00 CMD:0xC4 DEV:0x00 DEVCTRL:0x08 ERROR:0x01 FEATURE:0x00 LBAL:0xAF LBAM:0xDB LBAH:0x00
[72886ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0xE0 DEVCTRL:0x08 ERROR:0x01 FEATURE:0x00 LBAL:0x3F LBAM:0xC6 LBAH:0x00
[73887ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x92 LBAH:0x06
[74888ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x3F LBAM:0x00 LBAH:0x06
[77891ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x45 LBAH:0x00
[... 7 lines cut ...]
[90904ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0xCF LBAM:0x09 LBAH:0x09
[94908ms] DBG -- IDE regs: STATUS:0x00 CMD:0x30 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0xB0 LBAH:0x4F
[96910ms] DBG -- IDE regs: STATUS:0x00 CMD:0x30 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x45 LBAM:0xC6 LBAH:0x06
[100914ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0xCF LBAM:0x7E LBAH:0x07
[101915ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x4F LBAM:0x00 LBAH:0x09
[108922ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x0C
[109923ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x3F LBAM:0x98 LBAH:0x00
[... cut ...]
[124938ms] DBG -- IDE regs: STATUS:0x00 CMD:0x30 DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x06
[125939ms] DBG -- IDE regs: STATUS:0x00 CMD:0x30 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x7F LBAM:0xD5 LBAH:0x07
[126940ms] DBG -- IDE regs: STATUS:0x00 CMD:0x40 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x00
[127941ms] DBG -- IDE regs: STATUS:0x00 CMD:0x00 DEV:0x00 DEVCTRL:0x08 ERROR:0x01 FEATURE:0x00 LBAL:0x00 LBAM:0x00 LBAH:0x00
[129537ms] DBG IDE_EVENT_HWRST
```

morio, 2024-05-14:
@6ennadios I created a fix for another issue, I was wondering if you'd be will to try it?
https://github.com/rabbitholecomputing/ZuluIDE-firmware/actions/runs/9072485062

6ennadios, 2024-05-14:
This firmware works. Was able to run Dos games directly off of CD on my shareware disk and successfully installed and streamed Bad Mojo, an FMV heavy quicktime game.

[zululog.txt](https://github.com/rabbitholecomputing/ZuluIDE-firmware/files/15313512/zululog.txt)

closed 2024-05-15 by aperezbios