ZuluIDE/ZuluIDE-firmware #69: Upgrading firmware to v2024.05.03 results in corruption under MSDOS 6.22

jbehrends, 2024-05-12:
My setup:
- 386-40 / 16mb ram
- Motherboard with onboard single channel IDE controller 
- MSDOS 6.22 installed on a CF based drive as the Primary drive
- ZuluIDE connected as a Slave on the same ribbon cable as the CF based drive.
- MSDOS cdrom drivers:
  - config.sys: VIDE-CDD.SYS /D:ZuluIDE
  - autoexec.bat: SHSUCDX.EXE /D:ZuluIDE

Received my  ZuluIDE w/ firmware 2024.04.23-release
Tested with an iso of a cdrom image and everything worked perfectly! 
Upgrade the firmware to v2024.05.03-release
Tested with the same iso and "dir" returns a bunch of corruption.
Tried rebooting, power cycling, etc.  Still had corruption.
Rolled the firmware back to 2024.04.23-release
Everything works great once again!

Screenshot of running a "dir" on firmware v2024.05.03
[image]

Here are the debug logs for BOTH the 2024.05.03 and 2024.04.23 firmwares:

v2024.05.03
```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.05.03-release May  3 2024 19:58:03
[12ms] DIP switch settings: cablesel 0, drive_id 1 debug log 1
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2D5E51E3085062E6
[13ms] DBG Loading FPGA bitstream
[100ms] FPGA initialization succeeded
[116ms] SD card detected, FAT32 volume size: 30424 MB
[117ms] SD MID: 0x03, OID: 0x53 0x44
[117ms] SD Name: SC32G
[117ms] SD Date: 11/2022
[117ms] SD Serial: 0x937B0677
[166ms] Loading first image cdrm_LANCLNT.iso
[167ms] Device is a CDROM drive
[167ms] Loading image cdrm_LANCLNT.iso
[191ms] DBG Image file cdrm_LANCLNT.iso is contiguous, sectors 24640 to 546111
[198ms] Device 1 configuration:
[198ms] -- Max PIO mode: 3 (phy max 3)
[199ms] -- Max UDMA mode: 0 (phy max 0)
[199ms] -- Max blocksize: 4096 (phy max 4096)
[207ms] DBG -- Operating as secondary drive
[207ms] Initialization complete!
[209ms] DBG No Zulu Control board found, disabling 2nd core
[210ms] DBG IDE_EVENT_HWRST
[1211ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0xFF ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x51 0x5E 0x2D 0xFB 0xFF 0xFF 0x7F 0xA9 0xAF 0x9E 0xD6 0xEC 
[2000ms] FPGA license accepted with status 0x80
[10071ms] DBG IDE_EVENT_SWRST
[11072ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x55 LBAH:0xEB
[15076ms] DBG -- IDE regs: STATUS:0x00 CMD:0xEC DEV:0xB0 DEVCTRL:0x02 ERROR:0x01 FEATURE:0x00 LBAL:0x01 LBAM:0x56 LBAH:0x7C
[20258ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x01 LBAL:0x4A LBAM:0x71 LBAH:0x01
[22260ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x87 LBAL:0x42 LBAM:0x1A LBAH:0x00
[29267ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x4E LBAL:0xEC LBAM:0x0D LBAH:0x00
[30268ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0xF0 LBAL:0x4B LBAM:0x38 LBAH:0x00
[31269ms] DBG -- IDE regs: STATUS:0x00 CMD:0x20 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x00 LBAL:0x2A LBAM:0x91 LBAH:0x00
[32270ms] DBG -- IDE regs: STATUS:0x00 CMD:0x30 DEV:0xE0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0x12 LBAL:0x41 LBAM:0x00 LBAH:0x00
[33058ms] DBG IDE Command for DEV1: 0x08 IDE_CMD_DEVICE_RESET (device 0xB0, dev_ctrl 0x00, feature 0x87, sector_count 0x01, lba 0x00 0x5A 0xFF)
[33059ms] DBG -- Command complete
[33277ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x00 0x24 0x01)
[33278ms] WARNING: UltraDMA receive from IDE CRC mismatch, calculated 0x0000063F, host sent 0x0000
[33279ms] DBG -- ATAPI command: ATAPI_CMD_INQUIRY 0x12 0x00 0x00 0x00 0x24 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[33281ms] DBG ---- ATAPI send 1x36 bytes: 0x05 0x80 0x00 0x21 0x1F 0x00 0x00 0x00 0x5A 0x75 0x6C 0x75 0x49 0x44 0x45 0x00 0x43 0x44 0x2D 0x52 0x4F 0x4D 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x32 0x30 ... (total 36)
[33283ms] DBG -- ATAPI success
[33283ms] DBG -- Command complete
[33284ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x00 0x00 0x01)
[33285ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[... 100 lines cut ...]
[34510ms] DBG -- Command complete
[34511ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x08 0x00 0x2A)
[34512ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x00 0x10 0x00 0x00 0x01 0x00 0x00 0x00 
[34513ms] DBG -- Read 1 sectors starting at 16
[34514ms] DBG ---- Read CD: 1 sectors starting at 16, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 32768
[34517ms] DBG -- ATAPI success
[34517ms] DBG -- Command complete
[34525ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x08 0x00 0x2A)
[34526ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x00 0x16 0x00 0x00 0x01 0x00 0x00 0x00 
[34527ms] DBG -- Read 1 sectors starting at 22
[34527ms] DBG ---- Read CD: 1 sectors starting at 22, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 45056
[34532ms] DBG -- ATAPI success
[34532ms] DBG -- Command complete
[35534ms] DBG -- IDE regs: STATUS:0x40 CMD:0xA0 DEV:0xB0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0x00 LBAL:0x2A LBAM:0xFE LBAH:0xFF
[43814ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x00 0x00 0x2A)
[43815ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[43817ms] DBG -- ATAPI success
[43817ms] DBG -- Command complete
[87844ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x00 0x00 0x2A)
[87845ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[87847ms] DBG -- ATAPI success
[87847ms] DBG -- Command complete
```

Let me know if there is any additional information I can provide!

aperezbios, 2024-05-12:
@jbehrends thank you for the report. What's the motherboard/machine model of the computer this is happening on? I'd like to try and reproduce the issue.

jbehrends, 2024-05-12:
This is the exact motherboard I'm using:
https://theretroweb.com/motherboards/s/northgate-comput-elegance-sp
The onboard IDE controller of this board use a National Semiconductor Super IO chip model PC87310AVF to drive the IDE bus.  The chip is documented here:  https://theretroweb.com/expansion-chips/4479

If your curious, the cdrom iso I'm using is from OS/2 WARP 3.0... The exact iso can be found here:
https://archive.org/download/IBMOS2Warp3Collection
And the file that contains the iso is:
[IBM OS2 Warp 3 Connect - Blue - 8.200 - English - CDROM.zip](https://archive.org/download/IBMOS2Warp3Collection/IBM%20OS2%20Warp%203%20Connect%20-%20Blue%20-%208.200%20-%20English%20-%20CDROM.zip)

aperezbios, 2024-05-12:
@jbehrends could you try this firmware build and see if it results in the corruption problem not occurring? It will help narrow down which commit resulted in this regression
[https://github.com/rabbitholecomputing/ZuluIDE-firmware/actions/runs/8944008876/artifacts/1471761689](https://github.com/rabbitholecomputing/ZuluIDE-firmware/actions/runs/8944008876/artifacts/1471761689)

Please update via the .uf2 file.

jbehrends, 2024-05-13:
@aperezbios  I had a few minutes to spare, flashed the firmware using the .uf2 file.  The version in the debug log now shows  as:
```
[11ms] FW Version: 2024.04.29-devel May  3 2024 19:36:51
```

Did some quick testing and it appears to be working just fine.

Out of curiosity what was different in this firmware from the 2024.05.03 release?

aperezbios, 2024-05-13:
@jbehrends thank you for taking the time to test this build. We appreciate it. The build I had you test was the build immediately preceding [this commit](https://github.com/rabbitholecomputing/ZuluIDE-firmware/commit/8cb230431772e17d114ff17cd932bf4a9788a3d5), which made  a change to the FPGA code which resolved a different problem. In doing so, it appears we've caused this regression, but we don't know why yet.

morio, 2024-05-14:
@jbehrends could you try this build?
 https://github.com/rabbitholecomputing/ZuluIDE-firmware/actions/runs/9072485062 
It tries to fix the same issue that the new fpga bit stream addressed but not effect the timing as much.

jbehrends, 2024-05-14:
@morio  Flashed the firmware you provided and it appears that the cd image is still readable without errors.

```
[11ms] FW Version: 2024.05.04-devel May 14 2024 01:50:49
```

closed 2024-05-15 by aperezbios