ZuluIDE/ZuluIDE-firmware #77: Hangs boot with PC-9821As3

tdaede, 2024-06-06:
Installed as the primary (only) drive in a PC-9821As3, the machine's BIOS hangs while enumerating drives to boot from.

Debug log:

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.05.14-release May 14 2024 21:45:08
[12ms] DIP switch settings: cablesel 0, drive_id 0 debug log 1
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2C2578E3085062E6
[13ms] DBG Loading FPGA bitstream
[100ms] FPGA initialization succeeded
[116ms] SD card detected, FAT64 volume size: 60874 MB
[116ms] SD MID: 0x03, OID: 0x53 0x44
[116ms] SD Name: SN64G
[117ms] SD Date: 9/2023
[117ms] SD Serial: 0xxxxxxxxx
[132ms] Selecting device type when loading first image.
[133ms] Loading first image hddr0000.img
[134ms] Device is a hard drive
[134ms] Loading image hddr0000.img
[134ms] DBG Image file hddr0000.img is contiguous, sectors 66560 to 1115135
[142ms] Device 0 configuration:
[142ms] -- Max PIO mode: 3 (phy max 3)
[142ms] -- Max UDMA mode: 0 (phy max 0)
[143ms] -- Max blocksize: 4096 (phy max 4096)
[146ms] DBG -- Operating as primary drive, secondary drive not detected
[146ms] Initialization complete!
[148ms] DBG No Zulu Control board found, disabling 2nd core
[149ms] DBG IDE_EVENT_HWRST
[654ms] DBG -- Operating as primary drive, secondary drive not detected
[1150ms] DBG -- IDE regs: STATUS:0x40 CMD:0xFF DEV:0x00 DEVCTRL:0xFF ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x00 LBAH:0x00
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x78 0x25 0x2C 0xFF 0xFF 0xFF 0xFF 0x35 0x54 0x08 0xA4 0x76 
[2000ms] FPGA license accepted with status 0x83
```

I have a Rev2023e.

aperezbios, 2024-06-06:
@tdaede thanks for the report. As rigid hard drive support is still only experimental/in alpha, this isn't terribly surprising. Have you tried booting from a CD?

tdaede, 2024-06-08:
I tried using it as a CDROM drive (the machine can't boot from it), but with the same behavior. There is a bit more info in the logs though:

```
[661ms] DBG -- Config has_drive1=0, forcing second drive absence
[661ms] DBG -- Operating as primary drive, secondary drive not detected
[1156ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0xFF ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x78 0x25 0x2C 0xFB 0xFF 0xFF 0xFF 0x35 0x54 0x08 0xA4 0x76 
[2000ms] FPGA license accepted with status 0x83
[33726ms] DBG IDE_EVENT_SWRST
[33729ms] DBG IDE Command for DEV0: 0xEC IDE_CMD_IDENTIFY_DEVICE (device 0xA0, dev_ctrl 0x00, feature 0xAA, sector_count 0x01, lba 0xEB 0x14 0x01)
[33730ms] DBG -- Command complete
[33731ms] DBG IDE_EVENT_SWRST
[33734ms] DBG IDE Command for DEV0: 0xC6 IDE_CMD_SET_MULTIPLE_MODE (device 0xA0, dev_ctrl 0x00, feature 0xAA, sector_count 0x10, lba 0xEB 0x14 0x01)
[33735ms] -- Command handler failed for IDE_CMD_SET_MULTIPLE_MODE
[34737ms] DBG -- IDE regs: STATUS:0x41 CMD:0xC6 DEV:0xA0 DEVCTRL:0x00 ERROR:0x04 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
```

tdaede, 2024-06-08:
I should probably note that this is on the IDE bus intended for the HDD, so not entirely sure if the BIOS would like to see a CD-ROM here.

aperezbios, 2024-07-05:
> I should probably note that this is on the IDE bus intended for the HDD, so not entirely sure if the BIOS would like to see a CD-ROM here.

Agreed, in that case, this is probably expected behavior

closed 2024-09-05 by aperezbios

tdaede, 2024-09-05:
Has rigid device support improved? Should I retest this?

aperezbios, 2024-09-05:
It has, somewhat, but there is still more work to be done. A debug log from your ZuluIDE with the current firmware would be helpful, if you have the time, @tdaede

aperezbios, 2025-01-23:
@tdaede are you able to try this again with the latest [v2025.01.21](https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2025.01.21) firmware?