ZuluIDE/ZuluIDE-firmware #86: Eject-to-swap technique doesn't actually change the loaded image

mabalmer, 2024-07-17:
The method outlined to swap the loaded disk image described in the README doesn't appear to work with the newest firmware revision. Using a Pentium III-866 build running Windows XP SP3, and the ZuluIDE attached as the slave device on the primary IDE channel, I was able to get the device to successfully see one of several disc images on the SD card (strangely, the last image alphabetically), but ejecting the image through the OS simply causes the ZuluIDE to reload the exact same image.

aperezbios, 2024-07-17:
@mabalmer thanks for the report. Without a copy of your zululog.txt file, or, at a bare minimum, knowing the firmware version that's currently on your ZuluIDE, we can't do much with this report. Can you please attach the zululog.txt file from your SD card? Thanks :)

mabalmer, 2024-07-17:
Logfile attached. Debug switch was turned on.

[zululog.txt](https://github.com/user-attachments/files/16275021/zululog.txt)

morio, 2024-07-17:
@mabalmer could you try putting the prefix `cdrm` infront of your file names?

morio, 2024-07-17:
@mabalmer what are the other filenames for your other CD images? Before adding the cdrm prefix.

mabalmer, 2024-07-18:
Filenames on the card, alphabetically:

3com-install.iso
Stealth_II.iso
vb6.iso
Voodoo2Install.iso
Win98SE.iso
WinXPSP3.iso
WinXPSP3.iso

mabalmer, 2024-07-18:
Also, if there's a design (even if rudimentary) for the control board, I would love to know how to make it.

morio, 2024-07-19:
@mabalmer 
We do have a prototype design based on work from Tony over at PiSCSI
https://github.com/rabbitholecomputing/piscsi-hw/tree/main/zulu-ctrl-board-v1.0-kicad

It is what the current work on the controller code that Dustin (hoffman373) has been using.

The current code uses whatever SDFat brings up as the first image. When I get chance, it would make sense to push all the images through the alphabetic sorter first.

Lastly the image's first character must be alphanumeric so we use a period to disable images `.cdrm - disabled.iso` or use a subdirectory called `disabled` and move images to and from the directory.

mabalmer, 2024-07-19:
> @mabalmer could you try putting the prefix `cdrm` infront of your file names?

Just tried doing that. No dice -- this time, it loaded the Voodoo drivers CD, but again, _only_ the Voodoo drivers CD. Eject would simply cause it to "eject" the image and then reload it.

[image]

[zululog.txt](https://github.com/user-attachments/files/16316674/zululog.txt)

morio, 2024-07-24:
@mabalmer I'd forgotten we've moved to a new image selection mechanism and it doesn't have any alphabetic sorting. I'll try to add it back, it is part of the controller board code which I haven't really dealt with yet.

morio, 2024-07-25:
@mabalmer hopefully this sorts your ejection order alphanumerically https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/10101881581

aperezbios, 2024-08-06:
@mabalmer as we've not had a response from you in a couple of weeks, I'm going to close this issue, as it's been resolved for a couple of firmware updates now. Please upgrade to https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2024.08.02 when you have a chance.

closed 2024-08-06 by aperezbios