ZuluIDE/ZuluIDE-firmware #91: Add support USB storage

nvllsvm, 2024-07-30:
**What**
Add support for reading images and configuration from storage connected to the micro USB port. The supported filesystems and partition layout of the storage device would be expected to match that of the microsd. 

**Why**
One of my planned builds makes it difficult to mount the zuluide in a position where I can easily access the microsd slot. The case I'm using has two panel-mount USB ports on the front - one of which I would like to connect to the microusb port or some sort of headers on the zuluide-compact.

In another build, I'd like to use a pikvm to emulate a USB storage device. ZuluIDE supporting USB storage would then allow me to transfer contents using pikvm's web UI.

**Notes**
Someone seems to have added USB storage support to the ZuluSCSI. Not sure how much of that code could be reused here
https://68kmla.org/bb/index.php?threads/zuluscsi-rp2040-support-for-usb-mass-storage-access-sd-card-contents-via-usb.46039/.

aperezbios, 2024-07-30:
@nvllsvm thanks for your request. You're in luck, as this is actually already implemented, back on April 10th. The commit ID is 1bfcb61925f2c3b44f61996df7421f99a09263f8

I've added the missing usage instructions at https://github.com/ZuluIDE/ZuluIDE-firmware/wiki/USB-Mass-Storage-&-ZuluIDE, as well as a section in the main README file which points to it, along with an example directive in the zuluide.ini template itself.

If you have any questions, feel free to ask them here, but I'm going to close this issue for the time being, as this functionality already exists.

closed 2024-07-30 by aperezbios

nvllsvm, 2024-08-01:
Well this is embarrassing. The link I shared in the description isn't actually for the feature I'm looking for. What I'm specifically looking to do is have the zuluide act as a USB host and read images from a connected USB storage device - no the microsd.

aperezbios, 2024-08-01:
@nvllsvm in the interest of not making this already-closed issue a mess, I've moved your(new/different) request to its own issue,  #96, where I have some questions about use cases.