ZuluIDE/ZuluIDE-firmware #96: Add support for USB storage (zuluide acts as USB host)

aperezbios, 2024-08-01:
in another issue, @nvllsvm wrote "What I'm specifically looking to do is have the zuluide act as a USB host and read images from a connected USB storage device - not the MicroSD"

aperezbios, 2024-08-01:
@nvllsvm the USB controller in the RP2040 is limited to Full Speed (11Mbit/sec), or barely one megabyte per second, and does not support high speed USB. Given this hardware limitation, would such a feature still be of value? I'm not convinced it would be, but am open to other points of view on the matter. Can you elaborate?

nvllsvm, 2024-08-01:
> the USB controller in the RP2040 is limited to Full Speed (11Mbit/sec), or barely one megabyte per second

In that case, it's definitely not worth it. That would be roughly equivalent to CD ~8x and less than DVD 1x compared to the MicroSD speeds (CD ~48x , DVD ~4x). I'm fine with closing this issue.
 
---

For context on why I wanted this:
- one of my systems is bayless, but has two front-panel USB ports. Attaching one to the ZuluIDE would have been easier to work with since there is no way to extend the MicroSD port without case modification.
- another usecase would be using pikvm to emulate a USB mass storage guest so I can manage ZuluIDE's contents over a wired network connection.

closed 2024-08-01 by aperezbios