ZuluIDE/ZuluIDE-firmware #98: Windows 98 SE installation does not detect ZuluIDE as a CD drive

nvllsvm, 2024-08-06:
[zululog.txt](https://github.com/user-attachments/files/16515726/zululog.txt)
The Windows 98 SE (OEM) installation CD does not detect the ZuluIDE as a CD drive. The installation uses a universal driver which I've confirmed works with a Plextor PX-760A (released 2006) - amongst other drives. 

### Info

**Disc Image:** [Windows 98 Second Edition (OEM Full)](https://winworldpc.com/download/417d71c2-ae18-c39a-11c3-a4e284a2c3a5)
**Disc Image (sha256):** 2adfb46df8a9c7bbd2f67bff07461cc2f9d9ec8e01f0e112cb044c9e3e62f607
**Configuration:**
```
[IDE]
device = CDROM
atapi_vendor = "PLEXTOR"
atapi_product = "DVDR   PX-760A"
atapi_version = "1.07"
```

### Log
**Note:** The driver attempts to find a drive starting at **137651ms**
[zululog.txt](https://github.com/user-attachments/files/16515727/zululog.txt)

aperezbios, 2024-08-06:
@nvllsvm we're able to reproduce this in-house as well, so we will get it dealt with shortly.

morio, 2024-08-14:
@nvllsvm try this build. 
I was able to install your iso on a machine, but the machine isn't fully compatible with Win98 so I could only boot into Safe Mode.

aperezbios, 2024-08-14:
@nvllsvm we've decided to cut a new release which incorporates the above change, so please test with https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2024.08.14 instead.

nvllsvm, 2024-08-15:
I haven't tested a full installation, but can confirm that the Win 98 setup CD detects the ZuluIDE and begins setup.

closed 2024-08-15 by nvllsvm