ZuluIDE/ZuluIDE-firmware #109: Add support for audio output

dzlia, 2024-09-07:
Many retro gamers find valuable in a hardware CDROM emulator to have working audio output, so that games that rely on this feature work as expected. Lack of this feature makes this device useless for them.
Please consider possibility to add support for audio output in some form to cover this use case.

sererinn, 2024-09-11:
Same here, would see a lot of value in this and make it a complete product from my view.

JanSchr, 2024-09-11:
Is this something that could be added to the current hardware later (I saw two unpopulated 4 pin connectors on the board) or is a redesign of the hardware required?

Considering to buy a couple of Zulu's, would be cool if those could be re-fitted later on with the gui with menu and CD-Audio (if this ever get's developed).

aperezbios, 2024-09-11:
@JanSchr no hardware changes will be necessary, which is by design. 

We currently have a proof of concept implementation of CD audio output working, and are actively doing the necessary work to get the required firmware changes into a state where they can be merged. 

All that should be required hardware wise is an add-on board with an i2s DAC.

JanSchr, 2024-09-11:
Thanks for clearing this up, that's great news. Tried to buy one, but was unable to. :-( Rabbit hole on eBay does not Ship to Germany, Rabbit Hole does not list this device on there own site and the only listed Reseller for Germany is not listing it as well. Any ideas how to order this nice little thing from Germany? Would be appreciated.

aperezbios, 2024-09-11:
@JanSchr I double-checked, and there's nothing obvious in the eBay listing that should be preventing you from being able to order these from Germany. Can you please contact us via our website, so we can gather more information from you about the specifics of what you're encountering? Thanks!

JanSchr, 2024-09-11:
Did that, thank you very much for all the quick answers!

morio, 2024-10-02:
Audio support works as a beta.
We are testing a shield for the ZuluIDE that will have a common MPC-2 CD Audio header that will let you use a CDROM audio cable hooked into your sound card.
But if you want audio now you can purchase a https://www.amazon.com/dp/B07Q9K5MT8/
On the back of the ZuluIDE the GPIO pins you need to connect are labeled and should be connected to the DAC like so
```
ZuluIDE -> DAC
------------------
GND    -> GND
GPIO12 -> BCK
GPIO13 -> LRCK
GPIO14 -> DIN
```
Right next to the USB port on the ZuluIDE there is a pin labeled 5V that needs to be connected to VIN on the DAC

I was also able to use the 3.3V between GND and GPIO12 on the ZuluIDE for VIN on the DAC and it seemed to work fine.

Unfortunately the DAC board does not have the correct pinout for a CDROM audio cable as the header is on the DAC board is  Left Gnd Right Gnd and the CDROM audio cable is R G G L. I just used some female to female cables to hook the DAC directly into the sound card.
 
 You can find a uf2 binary with audio support here:
 https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11061395060
 
 And the current working branch here:
https://github.com/ZuluIDE/ZuluIDE-firmware/tree/feature/audio

Note: these links will get stale eventually but hopefully everything will be merged into main by then.

Here is a couple pictures of the unit I tested:

[image]
[image]

nvllsvm, 2024-10-05:
I can test this, though I'm curious - would it be possible to connect the ZuluIDE to the Digital CD audio input of a sound card instead?

aperezbios, 2024-10-06:
@nvllsvm not at this time, no. In theory, it would be possible to do so, but you would need to convert the i2s that's being generated by the RP2040 to SPDIF, instead of using a DAC.

nvllsvm, 2024-10-12:
I assmebled and tested this (commit https://github.com/ZuluIDE/ZuluIDE-firmware/commit/74f00a3ab66365b772ad9b159ab7e3209a8bd0ab). I tested the following on Windows 98 SE:
- music using the built-in CD Player program (multi-track audio-only CD)
- Quake 2's in-game music (mixed mode CD)

I also attempted to test Quake 1's audio when booting Windows 98 directly into DOS mode, but wasn't able to. Attempting to list the contents of the `D:/` hung and presented an error only when I removed the microsd from the ZuluIDE. I'll have to do some more testing to determine whether this is an error in system configuration (CONFIG.SYS, AUTOEXEC.BAT) or in ZuluIDE.

---

I should note that the DAC board I ordered from the link in https://github.com/ZuluIDE/ZuluIDE-firmware/issues/109#issuecomment-2389979904 did not have any of the pads bridged on the bottom. Only after soldering them did everything work.
[image]

sererinn, 2024-10-31:
Will there be any ZuluIDE cards coming pre-assembled with the necessary parts to produce CD Audio output? Cheers!

sererinn, 2024-10-31:
Thanks, this is great news. When that solution is here, I'll jump on it, I'm uncomfortable with soldering, quite a lot! :)

lucasstinis, 2024-10-31:
> Thanks, this is great news. When that solution is here, I'll jump on it, I'm uncomfortable with soldering, quite a lot! :) 

Same! As soon as there's a ready to go version that includes it I will order one immediately.

mabalmer, 2024-11-08:
> I also attempted to test Quake 1's audio when booting Windows 98 directly into DOS mode, but wasn't able to. Attempting to list the contents of the `D:/` hung and presented an error only when I [...]

I suspect this issue may have been related to #122 which was just addressed with today's firmware. DOS apparently didn't follow the logical method to determine if the drive was ready, so you might try updating to the 11.8.2024 build to see if that particular issue is fixed.

aperezbios, 2024-11-09:
@dzlia @JanSchr @nvllsvm and @lucasstinis the firmware implementation required to enable CD Audio output functionality is now present in ZuluIDE firmware [v2024.11.08](https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2024.11.08), which was released today. 

We have also developed an i2s DAC shield for existing ZuluIDE RP2040 Compact boards, which will be commercially available shortly. Closing as complete. The use of the i2s DAC shield we have developed is purely optional, as you can also make use of off-the-shelf DAC modules such as the one you can [obtain from Amazon](https://www.amazon.com/dp/B07Q9K5MT8/)

closed 2024-11-09 by aperezbios

aperezbios, 2024-11-20:
The [ZuluIDE i2s DAC shield upgrade kits ](https://shop.rabbitholecomputing.com/products/zuluide-cd-audio-dac-upgrade-kit-for-existing-zuluide-owners?)are now available for purchase from Rabbit Hole Computing. 

[image]

InterClaw, 2024-11-20:
Just to manage my expectations, this shield needs to be soldered to the ZuluIDE, right?

aperezbios, 2024-11-20:
@InterClaw It depends. If you already own a ZuluIDE, you'd need to solder the three black socket to the ZuluIDE board itself. At that point, the shield attaches to the ZuluIDE via the main board. You would get exactly what you see in the photos. 

If you do not already possess a ZuluIDE, all new purchases of ZuluIDE from Rabbit Hole Computing from now on come with the necessary sockets assembled by us.

If you don't have access to soldering equipment or are otherwise not confident enough about your soldering skills, we'd be happy to perform the upgrade for you, provided it makes economic sense to do so. You can contact us via our support e-mail address, support@zuluide.com and we will do what we can to assist.