ZuluIDE/ZuluIDE-firmware #115: Add support for multi-file cuesheets

nvllsvm, 2024-10-10:
It's valid for a cuesheet to reference multiple files (ex. separate files for each track). The disc preservation community [redump.org](redump.org) encourages this style when archiving physical media. All of my personal disc dumps are made in this style. However, I cannot use these dumps as-is with ZuluIDE.

However, ZuluIDE currently ignores .cue files when scanning for images. It will only read from a .cue file after initially finding a .bin file with the same stem. Even then, the files referenced in the .cue file are not utilitized - only the initial .bin is. This can and has resulted in ZuluIDE exposing an incomplete CD to the host.

It is possible to work around this by merging the separate files into a single bin (ex. `binmerge <split.cue> <merged_name>`), though I'm hoping that ZuluIDE could simply handle the split cuesheets instead.

aperezbios, 2024-10-11:
@nvllsvm acknowledged, and completely agreed on all points. This is estimated to be 10-20 hours worth of work, which is the only reason it isn't done yet.

PetteriAimonen, 2024-10-31:
There is now a test build available here: https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11616020341

The files have to be placed in a subfolder, for example:

     /Rise_of_the_Triad/Rise of the Triad - The HUNT Begins (USA).cue
     /Rise_of_the_Triad/Rise of the Triad - The HUNT Begins (USA) (Track 1).bin
     /Rise_of_the_Triad/Rise of the Triad - The HUNT Begins (USA) (Track 2).bin

nvllsvm, 2024-11-10:
Confirmed, thank you!

closed 2024-11-10 by nvllsvm

aperezbios, 2024-11-17:
Support for multi-file cuesheets is now implemented in the most recent [v2024.11.13](https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2024.11.13) firmware release