ZuluIDE/ZuluIDE-firmware #116: Add support for multisession CD images

nvllsvm, 2024-10-12:
Add support for multi-session CD images. I'm not sure what the most common or "best" format multisession images exist as, but redump's cuesheet style is already supported by cdemu.

Similar feature request for cdemu: https://github.com/cdemu/cdemu/issues/31

### Example 1: Duke Nukem - Music To Score By (first session audio, second session data) (i'll add a link once archive.org is back online).
```
REM SESSION 01
FILE "disc (Track 01).bin" BINARY
  TRACK 01 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 02).bin" BINARY
  TRACK 02 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 03).bin" BINARY
  TRACK 03 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 04).bin" BINARY
  TRACK 04 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 05).bin" BINARY
  TRACK 05 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 06).bin" BINARY
  TRACK 06 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 07).bin" BINARY
  TRACK 07 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 08).bin" BINARY
  TRACK 08 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 09).bin" BINARY
  TRACK 09 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 10).bin" BINARY
  TRACK 10 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 11).bin" BINARY
  TRACK 11 AUDIO
    INDEX 01 00:00:00
FILE "disc (Track 12).bin" BINARY
  TRACK 12 AUDIO
    INDEX 01 00:00:00
REM LEAD-OUT 01:30:00
REM SESSION 02
REM LEAD-IN 01:00:00
REM PREGAP 00:02:00
FILE "disc (Track 13).bin" BINARY
  TRACK 13 MODE2/2352
    INDEX 01 00:00:00
```

### Example 2: [redump - Carnivores (USA)](http://redump.org/disc/63778/) (both sessions are data)
```
REM SESSION 01
FILE "Carnivores (USA) (Track 1).bin" BINARY
  TRACK 01 MODE2/2352
    INDEX 01 00:00:00
REM SESSION 02
FILE "Carnivores (USA) (Track 2).bin" BINARY
  TRACK 02 MODE2/2352
    INDEX 01 00:00:00
```

nvllsvm, 2024-10-12:
Should be blocked by https://github.com/ZuluIDE/ZuluIDE-firmware/issues/115

aperezbios, 2024-11-14:
This is now implemented, and incorporated in to the v2024.11.13 firmware release

nvllsvm, 2024-11-26:
This doesn't seem to be resolved. To be clear, this is issue is for supporting CD images containing multiple sessions, not multiple files. Ex. https://en.wikipedia.org/wiki/Enhanced_CD

For example, the disc image for [Duke Nukem - Music To Score By](https://archive.org/details/duke_nukem_music_to_score_by) does not allow accessing the data track using ZuluIDE. The output of the Linux command `cd-info` shows a few key differences between the physical disc and using ZuluIDE.

```
<  13: 48:47:42  219417 data   false  no   
< 170: 68:09:30  306555 leadout (687 MB raw, 687 MB formatted)
< Media Catalog Number (MCN): 
< Last CD Session LSN: 219417
---
>  13: 46:15:42  208017 data   false  no   
> 170: 65:37:30  295155 leadout (662 MB raw, 662 MB formatted)
> Media Catalog Number (MCN): not supported by drive/driver
> Last CD Session LSN: 0
72c72
< session #2 starts at track 13, LSN: 219417, ISO 9660 blocks: 306403
---
> session #2 starts at track 13, LSN: 208017, ISO 9660 blocks: 306403                                                                   
```

[cd-info_plextor_px760a.txt](https://github.com/user-attachments/files/17912189/cd-info_plextor_px760a.txt)
[cd-info_zuluide.txt](https://github.com/user-attachments/files/17912190/cd-info_zuluide.txt)