ZuluIDE/ZuluIDE-firmware #123: Quake II stutters heavily when using .cue w/ multiple .bin

nvllsvm, 2024-11-10:
Quake II stutters heavily when playing audio from a `.cue` that references multiple `.bin` files. The stuttering disappears when using a `.cue` that references a single `.bin` (created with the binmerge command). 

**Disc**: [Quake II (USA)](http://redump.org/disc/2569/)

[zululog_merged.txt](https://github.com/user-attachments/files/17692393/zululog_merged.txt)
[zululog_split.txt](https://github.com/user-attachments/files/17692394/zululog_split.txt)

aperezbios, 2024-11-10:
@nvllsvm thanks for the report. We'll attempt to reproduce and resolve the issue in the coming week.

morio, 2024-11-13:
@nvllsvm could you try this build https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11809412087 
I'm not sure what happens when the track rolls onto the next one, but it seemed to take care of the stutter when I opened up Quake II with a multi-bin cue directory.

nvllsvm, 2024-11-13:
@morio That fixes the issue, thank you!

morio, 2024-11-13:
@nvllsvm thanks for the catch, https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11824493731 
this fixed it for me.

nvllsvm, 2024-11-14:
That fixed it for me too. Thanks!

closed 2024-11-14 by nvllsvm

aperezbios, 2024-11-14:
@nvllsvm these changes are now incorporated in to the [v2024.11.13](https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2024.11.13) release, which was cut earlier today.