ZuluIDE/ZuluIDE-firmware #124: Slow initial read when using VIDECDD.SYS in Win98 DOS mode

nvllsvm, 2024-11-10:
In Window 98's DOS mode, I'm noticing an usual performance difference between the OAKCDROM.SYS driver (from the Win98 boot floppy) and [VIDECDD.SYS](https://www.philscomputerlab.com/ms-dos-starter-pack.html). After booting to Win98's DOS mode, I navigate to `D:\` and run `dir`

- With OAKCDROM.SYS, `dir` executed immediately
- With VIDECDD.SYS, `dir` hangs for ~90 seconds

### Info
**Disc:** [Quake (USA)](http://redump.org/disc/43240/)

`CONFIG.SYS`
```
DEVICE=C:\WINDOWS\HIMEM.SYS /TESTMEM:OFF
DEVICE=C:\WINDOWS\EMM386.EXE RAM
DEVICEHIGH=C:\DRIVERS\OAKCDROM.SYS /D:OPTICAL
```
or
```
DEVICE=C:\WINDOWS\HIMEM.SYS /TESTMEM:OFF
DEVICE=C:\WINDOWS\EMM386.EXE RAM
DEVICEHIGH=C:\DRIVERS\VIDECDD.SYS /D:OPTICAL
```

`AUTOEXEC.BAT`
```
LH=C:\WINDOWS\COMMAND\MSCDEX.EXE /D:OPTICAL
```

morio, 2024-11-12:
@nvllsvm see if this fixes your issue
https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11788755304

nvllsvm, 2024-11-12:
@morio Confirmed fixed, thank you!

nvllsvm, 2024-11-12:
Reopening until the fix is merged to main

closed 2024-11-12 by nvllsvm