ZuluIDE/ZuluIDE-firmware #137: Repeated crash on hot swapping SDCard with Windows 98 complete freeze

dzlia, 2024-11-21:
I see repeatedly that Windows 98 hangs completely (no reaction on mouse movement) on hot SDCard hot swap. The green LED of ZuluIDE in the mean time is flashing with frequency ~1 Hz. Logs attached
[zuluerr.txt](https://github.com/user-attachments/files/17852040/zuluerr.txt)
[zululast.txt](https://github.com/user-attachments/files/17852041/zululast.txt)
[zululog.txt](https://github.com/user-attachments/files/17852042/zululog.txt)

dzlia, 2024-11-21:
Addition - before the system hangs completely it blinks as follows:
1 long blink
pause
1 short blink
short pause
1 long blink

aperezbios, 2024-11-21:
@dzlia both of the issues you filed seem to have the same root cause. 

In the attached Zululog.txt file, you can see "[58742ms] -- Host attempted read at LBA -134+1, beyond capacity 0x000000000004FB0E" and then the firmware crashes.

The real question for us to figure out is why this is happening.

dzlia, 2024-11-21:
If I can do something to simplify investigation then please let me know.
Currently the device is practically unusable for me.

dzlia, 2024-11-21:
Installation of WinQuake failed, too. It worked like this:
on copying ID1.PAK (large file) it stopped copying on every next 12%, than mouse hanged for second, then copying continued. On 75% it froze in the same way 2-3 times, then on 76, then on 92% and then on 99%, on copying GLquake.exe it started reporting I/O errors without trying to copy anything. Logs attached (they can contain previous attempts to use ZuluIDE, too, but I didn't do anything after failed installation of Quake). Hope this can help.
[zuluerr.txt](https://github.com/user-attachments/files/17852548/zuluerr.txt)
[zululog.txt](https://github.com/user-attachments/files/17852550/zululog.txt)

aperezbios, 2024-11-21:
@dzlia further log files aren't necessary at this time. According to the information you've shared already, it looks like the machine is an [IBM Intellistation M Pro type 6850](https://www.ibm.com/support/pages/overview-ibm-intellistation-m-pro-type-6850). Is this correct?

dzlia, 2024-11-21:
I've got a noname computer case with this motherboard. If any other hardware information can help then I can share it.

With the firmware of 08-Oct-2024 there is no difference in behaviour, except for freezes during WinQuake installation happening at different percents of progress - 4-16-27-43 - then I aborted it.

aperezbios, 2025-12-03:
@dzlia is this still an issue for you with the latest firmware?

Also, can you contact me via e-mail at support@zuluide.com? I have an idea I want to run by you.

aperezbios, 2026-04-27:
@dzlia can you try with the latest ZuluIDE firmware release, which you can download at https://github.com/ZuluIDE/ZuluIDE-firmware/releases/download/v2026.04.09/ZuluIDE_RP2040_2026-04-09.bin, and let us know if this is still an issue?

Thanks.