ZuluIDE/ZuluIDE-firmware #138: Data reading errors

dzlia, 2024-11-21:
First attempt to install something from ZuluIDE caused errors on reading files from an ISO file. Logs attached.

The SDCard is readable fine:
> diff -s /mnt/tmp/Oasis.iso Oasis.iso
Files /mnt/tmp/Oasis.iso and Oasis.iso are identical

[zululog.txt](https://github.com/user-attachments/files/17852266/zululog.txt)

aperezbios, 2024-11-21:
@dzlia thanks for the report. We will get this figured out as soon as we can. Can you please share with us the make and model of motherboard you have, assuming it's a commodity PC of some sort? This will help us replicate the problem in-house.

dzlia, 2024-11-21:
Attaching here AIDA32 reports related to the motherboard from this computer.
Just to add - it is a single device on the IDE controller.
[motherboard.txt](https://github.com/user-attachments/files/17852393/motherboard.txt)
[chipset.txt](https://github.com/user-attachments/files/17852394/chipset.txt)
[bios.txt](https://github.com/user-attachments/files/17852395/bios.txt)

aperezbios, 2024-11-21:
@dzlia  thanks. Based on this information, the machine has a VIA VT82C694X Apollo Pro133A Chipset and a VIA VT82C571 IDE controller. Stand by.

morio, 2024-11-21:
@dzlia could try this older firmware? https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2024.10.08
I'd like to rule out a recent clocking change.
Thanks!

dzlia, 2024-11-21:
@morio with ZuluIDE loaded with this firmware it looks like hot swapping does not crash the system anymore, or at least it is more stable and I cannot recreate the problem with 5 retries.

(Still, copying of files reported in #137 still fails)

dzlia, 2024-11-21:
Have just checked the logs and, despite of having no crashes, there is still that repeated
> [1449342ms] -- Host attempted read at LBA -134+1, beyond capacity 0x0000000000026109

[zululog.txt](https://github.com/user-attachments/files/17852945/zululog.txt)

aperezbios, 2024-11-21:
@dzlia with the older firmware, can you try setting the following in a zuluide.ini file:

```[IDE]
max_udma = -1```

This will disable negotiaton of Ultra DMA mode, restricting the ZuluIDE to PIO mode 3. This potential workaround may result in correct operation.

morio, 2024-11-21:
@dzlia could you turn debug on?
It is the bottom most rocker on the DIP switch. Then post your log files with both firmware versions.
Thanks

morio, 2024-11-22:
Try this firmware so you don't have to open your machine.
Just set 
```
[IDE]
debug = 1
```
in the zuluide.ini file
[zuluide-debug-via-sd-card.zip](https://github.com/user-attachments/files/17853581/zuluide-debug-via-sd-card.zip)

morio, 2024-11-22:
About the button, please contact support@zuluide.com for any hardware related issues

PetteriAimonen, 2024-11-22:
I think there is a lot of data corruption happening in the communication, which leads into incorrect commands being received and the CRC mismatches. The debug log could shed some light on it.

I would also suggest trying another IDE cable if possible. If you take the board out of the machine, try to check the soldering on the IDE connector and these components, just in case there happens to be more soldering issues with your board:
[image]

dzlia, 2024-11-22:
@morio, I tried this, but it doesn't seem to enable debug logging. Attaching a screenshot with the structure of files on the SD card and the log file. Do I need to push the bootloader button in order to activate zuluide-debug-via-sd-card.uf2 ?

[image]
[zululog.txt](https://github.com/user-attachments/files/17867846/zululog.txt)

dzlia, 2024-11-22:
@PetteriAimonen I successfully installed and used the system on this secondary controller, with exactly this IDE cable. When and if I extract uluIDE from the computer (which I prefer not to do because it is painful and also threads cut in the soft plastic case of ZuluIDE suffer from this) then I'll check and make photos of soltering around the IDE port and related chips. Before installation I also saw residues of flux on the back side of the molex port, so I cannot exclude possibility of other soldering defects.
However, errors appear with some pattern rather than absolutely randomly, which kind of points to software.
Anyway, with post PCB check results whenever I have a chance to conduct such a check.

PetteriAimonen, 2024-11-22:
I think we actually did not have `debug = 1` .ini setting for ZuluIDE (only ZuluSCSI).
I added it now in this firmware build: https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11969940522 so that you can enable debug without opening the enclosure.

I agree that it is not very probable that soldering defects would cause such a repeatable and late failure. Also the IDE cable can be considered good, as it works with another device.

Let's see about the debug log and then we can try to add more specific debugging to try to catch what is happening. It could also be some kind of timing issue.

This .ini file is worth trying (it forces a slower transfer mode):

     [IDE]
     debug = 1
     max_udma = -1
     max_pio = 1

dzlia, 2024-11-22:
@PetteriAimonen, I tried binaries from your reference. Unfortunately it caused the device to be not detected even by BIOS. Attaching log.
[zululog.txt](https://github.com/user-attachments/files/17870128/zululog.txt)

dzlia, 2024-11-22:
Actually, I've got the same error with the firmware of 08-Oct and max_udma = -1 only. See screenshot
[image]

dzlia, 2024-11-22:
In the meantime I disabled DMA on the driver level, and I stopped seeing freezes and errors. So it is something related to UDMA.

dzlia, 2024-11-22:
However there are:
1) still watchdog timeouts (and some more rare freezes)
2) CD audio stopped working. In DMA mode it worked with glitches with rewinding and switching to next/previous tracks, but still the win98 CD audio player was able to play audio from the Quake CD. Now it obviously does not work at all, with correspondent errors in the log.
[zululog.txt](https://github.com/user-attachments/files/17870392/zululog.txt)

PetteriAimonen, 2024-11-22:
@dzlia Thanks for the information. Unless Morio figures something out, I'll try next week to reproduce this problem with Windows 98 here. I don't currently have any VIA VT82 chipset IDE controllers available for testing, they seem somewhat rare, so I'm not sure if it will fail for me.

dzlia, 2024-11-22:
Some more input. With the ini file as below and the firmware ZuluIDE_RP2040_2024-11-22_f93177b I did the following:
1) in PIO mode:
1a) Listen to some WinQuake tracks - sound appeared, navidation to the previous track does shows in the player rubbish frozen numbers instead of incrementing counter from 00:00, while sound still works
1b) Copied PAK0.ID and PAK1.ID - no reported errors
1) in logs there are still LBA errors, but no watchdog errors

2) Turned on DMA and restarted the OS (hot reboot)
2a) copied from Oasis CD the setup file using windows exporer - freezes with hanged mouse, as before.
2b) in logs there are LBA errors and watchdog errors.

I still don't see any of the debug mode. Did you try to enable debug mode using this firmware? I'd be happy to repeat whatever is needed to enable it using the ini file. If this won't work then let me know and I'll get access to the DIP switches.

Logs:
[zululog.txt](https://github.com/user-attachments/files/17871001/zululog.txt)

File system setup + ini file:
[image]

morio, 2024-11-22:
Sorry I gave you the wrong update file initially. Copy the `ZuluIDE_RP2040_2024-11-22_26f9096.bin` image to your SD card from the following zip file: https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/11979477098/artifacts/2226230085 

When you power on the ZuluIDE the led will blink rapidly indicating the firmware is updating.

You should then see this version string  `FW Version: 2024.11.22-dev Nov 22 2024 20:15:53` in your `zululog.txt` on the second line. Debug should then work.

The .uf2 files require you the press the bootloader button down and the .bin files let you update via SD card.

dzlia, 2024-11-22:
Thanks @morio. Attaching two debug logs, one for PIO, one for DMA (both modes configured purely in the win98 driver). Testing scenario in both cases - 1) cold boot; 2) install Oasis; 3) power off. In the DMA mode installation failed, so when the installed suggested to abort or retry I chose abort.
I see a pause in messaging before watchdog timeout - this is probably the moment when the system was completely frozen.
In both cases I only defined "debug=1" in the ini file, no any UDMA/PIO settings.
Let me know if anything else is needed from me.

[zululog_pio.txt](https://github.com/user-attachments/files/17876562/zululog_pio.txt)

[zululog_dma.txt](https://github.com/user-attachments/files/17876561/zululog_dma.txt)

aperezbios, 2024-11-22:
@dzlia thanks for these two debug logs. In PIO mode, there are no watchdog timeouts, and things seem to be behaving properly. ZuluIDE currently supports PIO modes 0-3.

In the UDMA0 mode, we see the following error and subsequent watchdog timeout. Here's a snippet of the UDMA-enabled debug log. @PetteriAimonen does this stack trace help at all?
```
[116230ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x01, sector_count 0x01, lba 0xFF 0xFF 0x01)
[116231ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x11 0x39 0x00 0x00 0x10 0x00 0x00 0x00 
[116232ms] DBG -- Read 16 sectors starting at 4409
[116233ms] DBG ---- Read CD: 16 sectors starting at 4409, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 9029632
[131224ms] --------------
[131224ms] WATCHDOG TIMEOUT, attempting bus reset
[131225ms] GPIO states: out 0x00FE0003 oe 0x0207807F in 0x3DFF02FB
[131225ms] STACK 0x2003FC10:    0x1003EC14 0x3DFF02FB 0x200107F4 0x1002BF01
[131226ms] STACK 0x2003FC20:    0x1003EC14 0x00000000 0xD0000001 0x00000000
[131227ms] STACK 0x2003FC30:    0x200107F4 0x00000000 0x00000000 0x00000000
[131228ms] STACK 0x2003FC40:    0x00000000 0x1002DD75 0x00000000 0x00000010
[131229ms] STACK 0x2003FC50:    0x00000001 0x50200000 0x00000001 0x00000002
[131229ms] STACK 0x2003FC60:    0x000000D0 0xFFFFFFF9 0x00000000 0x2003FCF7
[131230ms] STACK 0x2003FC70:    0x00000001 0x00000000 0x20005361 0x200047DD
[131231ms] STACK 0x2003FC80:    0x20004548 0x21000000 0x00000000 0x50200000
[131232ms] STACK 0x2003FC90:    0x00000001 0x00000000 0x20418803 0x2003FCF7
[131232ms] STACK 0x2003FCA0:    0x00000000 0x2001257C 0x00000000 0x200047DD
[131233ms] STACK 0x2003FCB0:    0x000000FF 0x20004521 0x00000000 0x00000001
[131234ms] STACK 0x2003FCC0:    0x2003FCF7 0x00000000 0x00000000 0x00000000
[131235ms] STACK 0x2003FCD0:    0x2003FCEF 0x2003FCF7 0x00000000 0x2001257C
[131235ms] STACK 0x2003FCE0:    0x0000000E 0x20005375 0x00000000 0x2001257C
[131236ms] STACK 0x2003FCF0:    0x00000800 0x0000000E 0x00000800 0x1002375F
[131237ms] STACK 0x2003FD00:    0x2000CB50 0x0000000E 0x00000800 0x2000CB50
[131238ms] DBG atapi_send_data_async(): interrupted
[131238ms] DBG -- CD read failed, starting offset 9029632 length 16
[131239ms] DBG -- ATAPI error: 0x03 0x0000 (MEDIUM_ERROR)
```

morio, 2024-11-23:
@PetteriAimonen 
We have an older VIA south bridge that is failing all UDMA READ10s.
Here is a screen shot of a controller we know works
[image]
Here are two captures from the VIA controller
[image]
[image]
The shorter one might be an ATAPI Inquiry command as that returns 36 bytes and has the feature register set to 1 (udma mode on) for the Packet command 

The longer one is most likely a failed read10 as there aren't any other ATAPI packet commands in the log that send more than the Inquiry command besides the READ10.
The logic analyzer doesn't have enough depth to capture the READ10 command and the UDMA transfer but it is a CD with sector sizes of 2048 bytes so there should be at least 1024 cycles of the IORdy signal. 

The biggest difference I see is signaling at the end between the working IDE controller and the VIA. On the controller that works with the ZuluIDE the controller negates  HDMARDY(DIOR) on the last transition of DSTROBE(IORdy). On the VIA controller 
when HDMARDY is negated DSTROBE still one more transition left, the VIA host also emits a pause so maybe the DSTROBE transitions are too quick?

Note: some of the signals are inverted to the point of view PIO transfers so the waveforms are logic levels not signals.
Below are the raw captures from the Intronix LogicPort device.
[via-contoller-udma-crc-errors.zip](https://github.com/user-attachments/files/17877972/via-contoller-udma-crc-errors.zip)

PetteriAimonen, 2024-11-25:
@morio Thanks for the captures, I'll take a deep dive into them tomorrow.

mabalmer, 2024-11-30:
@PetteriAimonen I've got a couple single-board computers using the VT82C686B southbridge chip that seems to exhibit a similar issue to OP -- when the DMA mode drivers are enabled, the system responsiveness slows to a crawl, but _only_ while the disk is being accessed. When the DMA mode drivers are _not_ installed, responsiveness is fine. 

Here's the TRW page for the chipset the board uses: [VIA PM133T (VT8606 + VT82C686)](https://theretroweb.com/chipsets/682#docs)
Here's the TRW page for the board itself: [Axiomtek SBC82621 Rev. A6](https://theretroweb.com/motherboards/s/axiom-tech-axiomtek-sbc82621-rev_a6-rc)

I've got a logic analyzer here in my workshop. If you can tell me where to point probes, I may be able to capture something, although it might be difficult due to the extremely tight layout of that particular board. If the signals you're looking for are coming directly across the IDE bus I can probably squeeze a series of probes in-line with things.

PetteriAimonen, 2024-11-30:
@mabalmer The signals are on the IDE connector. But for a useful capture you'd need at least D0-D7 and all the control signals, plus preferrably SWO on the debug connector, at 50 MHz or better samplerate. So it exceeds the capabilities of cheapest logic analyzers.

mabalmer, 2024-11-30:
I've got a TLA 715 with an TLA7AA4 8GHz timing, 450MHz state module in it. I presume that's probably more than fast enough?

aperezbios, 2024-11-30:
@mabalmer yep that should be more than sufficient :)

mabalmer, 2024-11-30:
@PetteriAimonen OK, so just to make sure I'm grabbing everything: you _don't_ need the upper eight data bits? Also, where can I pick off SWO from the board?

PetteriAimonen, 2024-12-01:
@mabalmer Heh, yeah, that's a pretty fancy logic analyzer :)

Here is a diagram on the signals. The upper data bits are definitely useful, but if you run out of channels and have to leave something off, those would be the least important signals.

There is no need to include the DASP / PDIAG / CSEL signals as those are just for primary/secondary drive detection. RST is also not really useful, it gets pulses only after something has gone wrong (could be a useful trigger though).

[image]

Hooking up all the signals is a bit annoying, I have been using an IDE cable with 3 connectors and connecting off the middle one.
It's awesome if you can take a capture, but if it feels like too much work, I'll get to it eventually.

aperezbios, 2024-12-01:
It should be noted that the vast majority of ZuluIDE Compact boards boards in circulation as of today are Rev 2024a, which have a slightly different layout, although they are largely identical as far as everything goes on the IDC connector side, as this is a standardized pinout.

Here's a product photo of the Rev 2024a board. Because the pinout of the Cortex-M header at J301 is standardized, the SWO pin is in the same position on the connector block, but the adjacent J302 has been replaced with a Pico-specific debug header, which uses a three-pin JST connector, intended for use with the [official Raspberry Pi/Pico Debug probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html), which sells for around USD $12

[image]

PetteriAimonen, 2024-12-02:
I'm using a free-running 50 or 100 MHz sampling. I think there isn't any easily available synchronous clock on the board or in the cable. The IDE bus itself is partially asynchronous in design, and the board uses a 24 MHz oscillator that is multiplied by PLL internal to the chips.

mabalmer, 2024-12-05:
@PetteriAimonen Well, I've tried the board three separate times in both Windows XP and Windows 98, and none of the installs exhibits the behavior I was seeing earlier. Perhaps it was a conflict with another card that was causing the sluggish behavior. I will also note that, during the chipset driver installation, it _only_ asks about DMA mode in Windows 98 -- perhaps by the time XP came around, the driver had been rewritten to implement things better or some other hardware-level conflict had been resolved.

mabalmer, 2024-12-07:
OK. Here's what I was able to capture: I kept noticing that I would get acceptable readings from the TLA _before_ Windows would start, but then after it was booted (and the problem was evident) the TLA just kept seeing oscillating signals and it wouldn't actually capture anything. 

So, I cranked the sampling time down to 10ns (it was at 20ns before, so 50MHz) and finally got a good capture of the really long pauses and hitches that were happening. 

This was after Windows was booted, and I had just set the system to run Scandisk, just to keep the disk moving. The system is *not* accessing the ZuluIDE at all at the moment. I can try that next if that would be illustrative. I can also take that capture and zoom in to specific spots if desired. The entire capture was 32M samples so it's much longer than this, but maybe this will help. 
[image]

This is a snippet of what's coming across 
[image]

Let me know if there's anything more I can provide.

PetteriAimonen, 2024-12-07:
I would need the raw capture file (preferrably some standard format like VCD, CSV, BIN, if the program can export that) that contains at least one failed command execution in full.

dzlia, 2024-12-28:
any progress on this?

PetteriAimonen, 2024-12-29:
I have ordered some VIA hardware in hopes that it would reproduce the bug,
but it hasn't yet arrived.

PetteriAimonen, 2025-01-16:
Sorry for the long delay, now I finally have a VIA chipset (VT82C686B) motherboard I'm able to reproduce this problem with.

I see two problems:
* BIOS occasionally fails to detect ZuluIDE (low impact problem)
* When DMA is enabled in Windows 98 driver, ZuluIDE fails to communicate and log gets `atapi_send_wait_finish() data write timeout` messages.

The symptoms I see are not exactly the same as in your logs, but I'll start by debugging these and we'll see how it proceeds.

PetteriAimonen, 2025-01-16:
Ok, there is some light in this issue after all.

BIOS detection problems will be fixed by PR #164.

It appears that the DMA issues (at least the ones I currently get) are caused by incorrect handling of host-terminated DMA transfers.

Most IDE host controllers use the *"9.13.3 Pausing an Ultra DMA data-in burst"* method when their buffers fill up. This involves raising the `IOR` signal until they are ready to transfer more.

The VIA controller when used with Windows 98 driver instead does the terminating protocol, where `IOW` signal is also raised. After the termination the device should raise `DMARQ` to indicate there is more data available, but ZuluIDE is not doing this.

[image]

This appears like it shouldn't be particularly difficult to fix, but I guess I'll find out tomorrow when I try to do it :)

mabalmer, 2025-01-16:
@PetteriAimonen 
So, that's good that there's some light shed on the hiccups with the VT82C696 -- maybe it really does come down to the way the VIA drivers handled things. The next thing in my mind is determining which driver pack OP used. The _likely_ answer is the VIA 4-in-1, but it might get more tangled if OP used something like an IBM-provided driver pack. My bigger concern is that if the drivers aren't actually at fault here, the southbridge chip (VT82C596B) _is_ what's at fault, since it's what technically contains the IDE controller. 

Just in case, the datasheet: [VT82C596 Southbridge](https://theretroweb.com/chip/documentation/vt82c596b-663b93a6f2481415068784.pdf)

@dzlia 
When you set the system up, did you use the standard VIA 4-in-1 driver pack to set things up, or did you use something else? Also, just to be clear that the motherboard in question is the right one, I'm guessing your board is this: https://theretroweb.com/motherboards/s/ibm-aptiva-2178

That board could actually be in an Aptiva system, but it was also found in machines labeled as "IBM PC 300GL" as well.

PetteriAimonen, 2025-01-16:
As for me, I used whatever driver Windows 98 SE automatically detected (no extra drivers installed), except that I went and enabled DMA for the CD drive manually (it was disabled by default in device manager).
The problem didn't appear to occur with Debian Linux on the same hardware.

dzlia, 2025-01-16:
> [@dzlia](https://github.com/dzlia) When you set the system up, did you use the standard VIA 4-in-1 driver pack to set things up, or did you use something else? Also, just to be clear that the [...]

@mabalmer, yes I use VIA 4-in1 with DMA enabled at driver installation time. So, the IDE controller is covered by drivers from VIA, and for ZuluIDE the standard drivers from Windows 98 are used.

The motherboard I use at the moment is IBM 61H2578, revision 003. Here is how it looks like: https://www.ebay.com/itm/404688733184. (As far as I know Alex Perez ordered a month ago a motherboard with the same part number, revision 002).

aperezbios, 2025-01-16:
@dzlia I can confirm that we have that motherboard, and @morio has been actively using it for testing.

mabalmer, 2025-01-16:
@dzlia Got it. Yeah, that's the driver/board combo I thought it would be. That it exhibits problems like it did with my setup as well seems to point towards half-baked drivers from VIA, but we'll see what can be ironed out. Thanks for clarifying.

PetteriAimonen, 2025-01-17:
@dzlia I believe this test build should fix the problems:
https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/12828870926/artifacts/2446648688

closed 2025-01-17 by aperezbios

dzlia, 2025-01-17:
@PetteriAimonen, looks like now there is an improvement in how data is read. I was able to install winquake with only one hiccup at 4%, and not sure if it is an error read or not (no errors in zululog.txt)

However, now in DMA mode only (Win 98 SE) eject (both from menu and hardware button eject) don't want to switch images normally. In PIO it is smooth as usual, in DMA it either doesn't do anything, or resets the CD root directory to be empty, or (there was at least twice) corrupted file system is displayed. One time it reported something like 12TB in ~7900 files for a mounted ISO image. In the log file switched between images are recorded, when there are no switched in Windows Explorer.
So, there is something still to improve w.r.t. to DMA operation. At the moment it is unusable.

I can create a separate bug if needed.

aperezbios, 2025-01-17:
@dzlia Thanks for testing. Please create a separate bug, as this appears to be a different problem.

dzlia, 2025-01-20:
@PetteriAimonen, it looks like that after uploading the proposed firmware I have broken play Audio CD functionality, even in PIO mode. Without anything in logs, playing tracks with Windows 98 CD player and from games don't work anymore (while I see the track lists in the CD player), and I've got two complete system lockups, both time with ZuluIDE LED lighting constantly.
I'll check latest stable firmware whether or not it still supports CD Audio.

aperezbios, 2025-01-20:
@dzlia the CD Audio playback regression has been confirmed, and is being tracked in issue #170, which you opened.

PR #167 (2025-01-17): FPGA: Support splitting one UDMA sector into multiple bursts
VIA 82C686 terminates UDMA transfers in the middle of a sector when its buffer fills up, and expects the device to continue. Previously ZuluIDE would hang in this situation.

Now DMARQ is reasserted if more data remains.
FPGA automatically combines the CRC checksums from all bursts of a single sector.

Tested with VIA drivers 1999 (included with Windows 98 SE), 2001 (downloaded from [retroweb](https://theretroweb.com/drivers/239)) and with Linux kernel 6.1.0.

Fixes #138