ZuluIDE/ZuluIDE-firmware #141: ZuluIDE not detected when microsd contains a lot of bin/cue images

nvllsvm, 2024-12-08:
**What**
The host PC does not detect the ZuluIDE where the microsd contains a large number of images. The ZuluIDE is not detected by the BIOS nor Windows 98.

In this instance, my microsd has the following contents (~70GB, see [file_info.json](https://github.com/user-attachments/files/18053719/file_info.json) for specifics and hashes):
- 120 folders containing a single `.cue` file and one or more `.bin` files
- 3 `.iso` files

Although I have an image set in `zululast.txt`, the ZuluIDE appears to be scanning and checking each `.cue` it finds before initializing CD emulation. Notably, there are 114 log lines containing `Failed to fetch bin/cue size` consuming the time between  2710ms and 55310ms:
```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.12.06-release Dec  6 2024 19:53:01
[12ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2DA350E3085062E6
[100ms] FPGA initialization succeeded
[100ms] I2S audio to expansion header enabled
[101ms] -- System clock is set to 135428571Hz
[107ms] SD card detected, FAT32 volume size: 244419 MB
[107ms] SD MID: 0x1B, OID: 0x53 0x4D
[108ms] SD Name: JE4S5
[108ms] SD Date: 6/2023
[108ms] SD Serial: 0xB2536349
[135ms] Selecting device type when loading first image.
[2710ms] Failed to fetch bin/cue size.
...
[55310ms] Failed to fetch bin/cue size.
[55728ms] Device is a CDROM drive
[55733ms] Device 0 configuration:
[55733ms] -- Max PIO mode: 3 (phy max 3)
[55734ms] -- Max UDMA mode: 0 (phy max 0)
[55734ms] -- Max blocksize: 4096 (phy max 4096)
[55740ms] Initialized platform with device control.
[55741ms] Set SSID from INI file to ---
[55742ms] Set PASSWORD from INI file.
[55743ms] Hardware UI not found.
[55743ms] I2C server found
[55743ms] Initialized platform controller with the status controller.
[55746ms] Initialized platform controller with the display controller.
[55747ms] Initialized platform controller with input receiver.
[57162ms] Failed to fetch bin/cue size.
[57162ms] Loading last used image: "Recoil"
[57163ms] Loading image "Recoil"
[57243ms] ---- Cue sheet Recoil (USA).cue loaded with 5 tracks
[57344ms] Initialization complete!
[57346ms] Hardware UI not found.
[57346ms] I2C server found
[57358ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x50 0xA3 0x2D 0xFF 0xFF 0xFF 0xFD 0x17 0x73 0xB3 0x4C 0x1B 
[57359ms] FPGA license accepted with status 0x81
[57429ms] I2C Client subscribed to updates.
[62986ms] Client IP address is: 192.168.5192.168.5.10
[98494ms] WARNING: Detected supply voltage drop to 2986mV. Verify power supply is adequate.
```

[zululog.txt](https://github.com/user-attachments/files/18053720/zululog.txt)

aperezbios, 2024-12-08:
@nvllsvm thanks for the report. Can you let us know if this also happens when the display and status controllers are NOT physically connected to the ZuluIDE? This information may help us narrow down the code path. If the issue does not manifest with both connected, please try using just one or the other, and let us know if it is usable.

nvllsvm, 2024-12-08:
The issue is the same after removing the DAC/wifi addon and deleting `zuluide.ini`:

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.12.06-release Dec  6 2024 19:53:01
[12ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2DA350E3085062E6
[100ms] FPGA initialization succeeded
[100ms] I2S audio to expansion header enabled
[101ms] -- System clock is set to 135428571Hz
[107ms] SD card detected, FAT32 volume size: 244419 MB
[107ms] SD MID: 0x1B, OID: 0x53 0x4D
[108ms] SD Name: JE4S5
[108ms] SD Date: 6/2023
[108ms] SD Serial: 0xB2536349
[178ms] Selecting device type when loading first image.
[2745ms] Failed to fetch bin/cue size.
...
[55332ms] Failed to fetch bin/cue size.
[55749ms] Device is a CDROM drive
[55819ms] Device 0 configuration:
[55819ms] -- Max PIO mode: 3 (phy max 3)
[55819ms] -- Max UDMA mode: 0 (phy max 0)
[55820ms] -- Max blocksize: 4096 (phy max 4096)
[55900ms] Initialized platform with device control.
[55921ms] Hardware UI not found.
[55921ms] I2C server not found
[57402ms] Failed to fetch bin/cue size.
[57402ms] Loading last used image: "Powerslave (USA)"
[57402ms] Loading image "Powerslave (USA)"
[57508ms] ---- Cue sheet Powerslave (USA).cue loaded with 19 tracks
[57641ms] Initialization complete!
[57643ms] Hardware UI not found.
[57643ms] I2C server not found
[57653ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x50 0xA3 0x2D 0xFA 0xFF 0xFF 0xFD 0x17 0x73 0xB3 0x4C 0x1B 
[57654ms] FPGA license accepted with status 0x80
```
[zululog.txt](https://github.com/user-attachments/files/18053815/zululog.txt)

nvllsvm, 2024-12-08:
The issue seems to be in the logic of determing what kind of device ZuluIDE should emulate. Setting `device = CDROM` in `zuluide.ini` eliminates the problem and the 114 log lines of `Failed to fetch bin/cue size.`.

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2024.12.06-release Dec  6 2024 19:53:01
[12ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[13ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2DA350E3085062E6
[100ms] FPGA initialization succeeded
[100ms] I2S audio to expansion header enabled
[101ms] -- System clock is set to 135428571Hz
[107ms] SD card detected, FAT32 volume size: 244419 MB
[107ms] SD MID: 0x1B, OID: 0x53 0x4D
[108ms] SD Name: JE4S5
[108ms] SD Date: 6/2023
[108ms] SD Serial: 0xB2536349
[154ms] Device is a CDROM drive
[165ms] Device 0 configuration:
[165ms] -- Max PIO mode: 3 (phy max 3)
[166ms] -- Max UDMA mode: 0 (phy max 0)
[166ms] -- Max blocksize: 4096 (phy max 4096)
[180ms] Initialized platform with device control.
[182ms] Set SSID from INI file to ---
[184ms] Set PASSWORD from INI file.
[184ms] Hardware UI not found.
[185ms] I2C server found
[185ms] Initialized platform controller with the status controller.
[188ms] Initialized platform controller with the display controller.
[188ms] Initialized platform controller with input receiver.
[665ms] Loading last used image: "Powerslave (USA)"
[665ms] Loading image "Powerslave (USA)"
[781ms] ---- Cue sheet Powerslave (USA).cue loaded with 19 tracks
[912ms] Initialization complete!
[914ms] Hardware UI not found.
[914ms] I2C server found
[997ms] I2C Client subscribed to updates.
[2000ms] FPGA license request code: 0xE6 0x62 0x50 0x08 0xE3 0x50 0xA3 0x2D 0xFB 0xFF 0xFF 0xFF 0x3E 0x23 0x2C 0x2E 0x3F 
[2000ms] FPGA license accepted with status 0x81
[6300ms] -- Set PIO transfer mode 3
[6311ms] -- Set UDMA transfer mode 0
[12055ms] Client IP address is: 192.168.5.10
[40992ms] WARNING: Detected supply voltage drop to 2986mV. Verify power supply is adequate.
```

There might be a related issue when listing images in the web ui, but I need to do some more testing to confirm.

hoffman373, 2024-12-08:
I have an idea on this and will create  test image tonight

hoffman373, 2024-12-09:
In branch fix/large-image-init I temporarily disabled reading the image size. Are you able to build your own images or would you like me to build and send you one to test?

I am going to create an SD card with a large number of images closer to your test case for my own testing but that may take a bit.

aperezbios, 2024-12-09:
@hoffman373 and @@nvllsvm the CI-built binaries for this branch can be downloaded from https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/12227959418/artifacts/2291561595

nvllsvm, 2024-12-09:
While that did eliminate the `Failed to fetch bin/cue size.` errors, it did not fix the slowness. I'm still looking into that. 

As for a proper fix to the size error, these two PR's fix it for me:
- https://github.com/ZuluIDE/ZuluIDE-firmware/pull/142
- https://github.com/rabbitholecomputing/CUEParser/pull/3

nvllsvm, 2024-12-10:
This fixes the slowness by auto detecting .cue files as a cdrom device.