ZuluIDE/ZuluIDE-firmware #149: Hardware eject button doesn't eject.

dzlia, 2025-01-02:
When I try to use the SMD switch button near the activity LED it doesn't seem to do anything. Literally, it doesn't eject the disc.

Conditions:
- device revision: 2024a
- firmware: 2024.12.10
- zuluide.ini
```
[IDE]
debug = 1
enable_usb_mass_storage = 1
eject_button = 1
```
- OS - Windows 98 SE
- access mode - PIO (forced by OS CDROM driver)
- log file recorded for a session when I pushed the button several times - attached.

What should happen now when this button is pushed?

As an additional observation: debug = 1 doesn't seem to do anything in the latest firmware.

[zululog.txt](https://github.com/user-attachments/files/18287062/zululog.txt)

morio, 2025-01-03:
Thanks for the catch.
Please try this build https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/12591174957
Tested locally on Windows 98 and Windows 8 which have different eject communication mechanisms.
This should also enable the button without any zuluide.ini settings.
Cheers!

dzlia, 2025-01-03:
With the advertised firmware, on first attempt the computer was locked up at post, and ZuluIDE's led blinked slowly. No crash in the log, but the log shows that something bad happened to ZuluIDE, because logging stopped in the middle of initialisation.

dzlia, 2025-01-03:
Actually, after very long pause, made of probably 20 blinks of the led, the BIOS has just made the device invisible and booted my Windows 98, where I don't see the device at all.

aperezbios, 2025-01-03:
@dzlia this issue is now resolved with the release I just made, which can be downloaded from https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2025.01.03

I can confirm there was a bug with the previous build that we asked you to try, which prevented correct hardware initialization when no hardware UI board was attached.

dzlia, 2025-01-04:
Yes, thanks. Now pushing the button eject on the device causes rotation among images on my computer with Windows 98, just like if the current CD was ejected using the Windows' context menu.

closed 2025-01-04 by aperezbios