ZuluIDE/ZuluIDE-firmware #152: USB connection not usable

dzlia, 2025-01-04:
### Discussed in https://github.com/ZuluIDE/ZuluIDE-firmware/discussions/136

<div type='discussions-op-text'>

<sup>Originally posted by **dzlia** November 21, 2024</sup>
I've plugged my ZuluIDE with the SDCard that you provided to a computer and started using it. However, an attempt to connect to the SDCard using the USB connection failed. All the time it is a) in Linux: fdisk /dev/sdb -> no medium found (on two different computers); b) in Windows 98 with USB stick drivers - mass storage device appears with zero capacity.
The device with the latest firmware showed contents of a manually copied ISO, and I see some logs on the SDCard, so the device is not a brick.

I tried this:
1) firmware it came with - 20-Aug-2024
2) firmware of 13-Nov-2024
3) connecting it by USB when the computer it is connected to by IDE and molex is on
4) connecting it by USB when the computer it is connected to by IDE and molex is off
5) connecting it by USB to the host computer using USB 1.1
6) connecting it by USB to the host computer using USB 2.0

All the time the result is the same.
And there is nothing important in logs with the latest firmware (while when I first detached the SDCard to check if it works at all I saw a zuluerror.txt on the file system, which have never appeared again with a new firmware).
</div>

aperezbios, 2025-01-06:
@dzlia This was a tricky one to track down, but we finally got there. Thanks for hanging in there. This is resolved with pull request #155  

[ZuluIDE-firmware-3f6a8bb-fix-usb-mass-storage.zip](https://github.com/user-attachments/files/18324677/ZuluIDE-firmware-3f6a8bb-fix-usb-mass-storage.zip)

dzlia, 2025-01-08:
@aperezbios, I've checked this firmware, and the mass storage exposure function worked for me in the basic form. I'll create a followup issue with a proposal how to make it better.

In addition, all the time I need to insert a micro SD card into ZuluIDE (which is in a computer case which is under the table) it's really pain in ass. This time when I tried to insert the card, it slipped over the port and felt on the ZuluIDE PCB, which can be even dangerous for the device and/or the SD card. I would recommend you in further revisions of the device to consider switching at least to a normal-size SD card port, which is way easier to use. It would be even better if you make a normal front panel for the device, so that it's internals are hidden.
It could be less of importance if the device is on a desk in the development/experimentation mode all the time, but when it is used as a substitution of a CDROM in a real computer it becomes more valuable.

closed 2025-01-17 by aperezbios