ZuluIDE/ZuluIDE-firmware #158: Possibility for LED to be off when no SD card.

dzlia, 2025-01-07:
When the drive doesn't have an SD card, it's LED flashes to indicate that the card is not here. If no SD card is expected (simply because CD-ROM is not used) it becomes annoying, especially at night. At the same time, the usefulness of this signal is also obvious. However, whether or not LED should blink on no SD card can be defined by the configuration. Could you consider adding this configuration, please?

aperezbios, 2025-01-15:
At the present time, it's not possible to have the LED activity be configurable if there is no SD card in the ZuluIDE, since the SD card itself the only existing mechanism we have for storing persistent settings. Without implementing an additional/secondary mechanism for storing settings within flash on the ZuluIDE, implementing this as a configurable option is not currently possible. 

We had an internal discussion about what it might take to implement functionality like this, and agreed that, for User Experience reasons, it would not be acceptable to fully disable the LED from flashing five times, however an acceptable compromise I came up with that I think would work for you would be to stop blinking five times after a pre-configured, or potentially user-configured, number of flash cycles. This would ensure that the board does not appear "dead" if it's power cycled.

We may circle back to this in the future, but for now, I'm closing this as wontfix.

dzlia, 2025-01-17:
@aperezbios it is perfectly fine for the LED to flash for some time at the boot time, to report the process of initialisation. But after the device has been initialised, it is usable to be able to suppress flashing if there is no CD and it is on purpose.
This is exactly how regular CD-ROM drives work and it is very handy.