ZuluIDE/ZuluIDE-firmware #160: Make possible writing to the SD card from the computer where ZuluIDE is connected.

dzlia, 2025-01-08:
As a follow up for the issue #152, here is a scenario of using USB mass storage connection of ZuluIDE that didn't work for me, but which could be important.

Imagine that the computer where ZuluIDE is plugged in is the only one around, or it is inconvenient to use any other computer (e.g. because it doesn't have micro SD card reader, or because moving around the micro SD card is not desirable), and something needs to be changes in the SD card, say, a new image copied from the hard drive, or some unneeded images need to be deleted/renamed, or configuration should be changed.

What happens when I plug the USB cable to the host (Windows 98) and ZuluIDE is that both the CDROM is mounted (because it cannot be unmounted) and also the SD card is attached as a removable disk. I can even read data from the disk. But an attempt to write anything to the SD card is rejected.
The only way to write to the SD card is to either use a different computer nearby and attach it to ZuluIDE when retro computer is powered off, or to plug out the SD card to write to it from somewhere else.

It would be way more convenient if one of the following works:
1. It is possible to unmount the active CD image even if the SD card is in ZuluIDE. And when no any CD is mounted then mounting the SD card as a USB mass storage device works in the write mode.
OR
2. It is possible to mount the SD card as a USB mass storage device in parallel to a mounted CD image, admitting that there could be a data race on accessing the same CD image and/or the zulu log files. This limitation in not a problem in most real cases.

aperezbios, 2025-01-15:
@dzlia  This functionality cannot be implemented as requested. As far as I'm aware, not possible because there is no way to safely allow writes from two different sources at the same time, as it's not possible to ensure filesystem integrity when doing so. Corruption would be the likely result. @PetteriAimonen feel free to chime in here if you have any further comments or explanations.

closed 2025-01-15 by aperezbios

PetteriAimonen, 2025-01-16:
I think this would be possible to do when the CD image is ejected from ZuluIDE (option 1). This would be the same as when the SD card is removed from ZuluIDE, it can show as empty drive. It is a usability question what would be the best way to activate it - should the drive be automatically ejected if USB write is attempted, or should user first press eject button.

I agree that simultaneous access (option 2) is a recipe for hard to identify corruption problems.

dzlia, 2025-01-17:
Having an option to eject an image and hold on with mounting another one can have value itself (you can check how it is done for instance in flashfloppy for an emulator of a floppy drive). And if it is combined with allowing to connect in the write mode by USB then it should be a perfect combo.