ZuluIDE/ZuluIDE-firmware #168: Switch between images doesn't work.

dzlia, 2025-01-17:
The fix proposed in #138 improved reading data, however, now in the DMA mode (Win 98 SE) eject (both from menu and hardware button eject) don't switch images normally. In the PIO mode it is smooth as usual, in DMA it either doesn't do anything, or resets the CD root directory to be empty, or (there was at least twice) leads to a corrupted image mounted. One time it was with file names like _______,  another time the CD root directory reported something like 12TB in ~7900 files for a mounted ISO image.
In the log file switches between images are recorded, while there are no normal switches between images in the operating system.
So, there is something still to improve w.r.t. to DMA operation. At the moment it is still unusable in my setup.

System: Windows 98 SE + DMA + motherboard IBM 61H2578, revision 003 (VIA Apollo 133A) + Via 4-in-1 drivers for the chipset.

morio, 2025-01-18:
@PetteriAimonen I found the symptom but not the cause

These are captures using an app bus hound 2.0 in windows 98 se.
[cdrom-eject-button.txt](https://github.com/user-attachments/files/18462464/cdrom-eject-button.txt)
[zuluide-dma-eject.txt](https://github.com/user-attachments/files/18462465/zuluide-dma-eject.txt)
[zuluide-pio-eject.txt](https://github.com/user-attachments/files/18462466/zuluide-pio-eject.txt)

SNS is sense data:
In pio mode we return
```
0:0:0:0   SNS  f0 00 02 00  ....             6us         4.3.0        
               00 00 00 0b  ....                         4.3.4   
```
Which is set here:
https://github.com/ZuluIDE/ZuluIDE-firmware/blob/4e01e0574721d7eb982908c33b0553298cdad214/src/ide_atapi.cpp#L1000-L1002

For some reason in UDMA mode these four bytes are 0x01 0x00 0x01 0x00 is sent before the response code (0xf0) and sense key (0x02 - not ready)
```
0:0:0:0   SNS  01 00 01 00  ....             7us         2.3.0        
               f0 00 02 00  ....                         2.3.4        

```

The rest of the sense data isn't shown because it is the freeware version of the app and only records 8 bytes of data.

An actual CD-ROM drive which I hooked up doesn't set the valid bit for the first byte of the sense data which is why it looks like this

```
0:0:0:0   SNS  70 00 02 00  p...             7us         3.3.0        
               00 00 00 0a  ....                         3.3.4        
```
where we do set the valid bit so we send 0xf0 where the cdrom sends 0x70.

PetteriAimonen, 2025-01-20:
For some reason I'm not getting this problem with my setup.
Can someone provide a debug log from ZuluIDE when this happens?

Here is also a test build that I made off the guess that it would have something to do with split UDMA transfer not getting terminated on error:
https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/12864229899

morio, 2025-01-20:
@PetteriAimonen  Here is the log for commit c687a53 where eject fails in Windows 98se. The log shows that the eject mechanism is switching the images. But on the OS side it isn't getting the correct sense data (.
[zululog-c687a53-eject-fails-dma-mode.txt](https://github.com/user-attachments/files/18481987/zululog-c687a53-eject-fails-dma-mode.txt)

In the log, the MCU is sending the proper sense data
`[79260ms] DBG ---- ATAPI send 1x14 bytes: 0xF0 0x00 0x02 0x00 0x00 0x00 0x00 0x0B 0x00 0x00 0x00 0x00 0x3A 0x00` 
from the below section.
```
[77245ms] DBG Ejection button pressed
[77248ms] Device ejecting media: "Quake2"
[78687ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[78689ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[78722ms] DBG Image file win98qi_v0.9.5_ALL.iso size 1907148800
[78889ms] DBG Image file win98qi_v0.9.5_ALL.iso is contiguous, sectors 4210752 to 7935679
[78890ms] Loading image "win98qi_v0.9.5_ALL.iso"
[78896ms] DBG Image file win98qi_v0.9.5_ALL.iso size 1907148800
[79063ms] DBG Image file win98qi_v0.9.5_ALL.iso is contiguous, sectors 4210752 to 7935679
[79076ms] DBG Image file win98qi_v0.9.5_ALL.iso size 1907148800
[79243ms] DBG Image file win98qi_v0.9.5_ALL.iso is contiguous, sectors 4210752 to 7935679
[79244ms] -- Device loading media: "win98qi_v0.9.5_ALL.iso"
[79253ms] DBG -- ATAPI error: 0x02 0x3A00 (NOT_READY)
[79253ms] DBG -- Command complete
[79258ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x01, sector_count 0x01, lba 0xFF 0xFF 0x01)
[79259ms] DBG -- ATAPI command: ATAPI_CMD_REQUEST_SENSE 0x03 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[79260ms] DBG ---- ATAPI send 1x14 bytes: 0xF0 0x00 0x02 0x00 0x00 0x00 0x00 0x0B 0x00 0x00 0x00 0x00 0x3A 0x00 
[79261ms] DBG -- ATAPI success
[79261ms] DBG -- Command IDE_CMD_PACKET completed with error status 0x24
[79263ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[79264ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[79265ms] DBG -- ATAPI success
[79265ms] DBG -- Command complete
```

You probably already do this, but to get the CD-ROM in DMA mode you have to check this box and restart.
[image]

Commit 4950dbb effectively breaks DMA mode for Win98se, as the OS does not keep the DMA setting after restarting the computer.
[zululog-commit-4950dbb-dma-will-not-stick.txt](https://github.com/user-attachments/files/18482014/zululog-commit-4950dbb-dma-will-not-stick.txt)

PetteriAimonen, 2025-01-21:
@morio Interesting. These problems do not appear to occur for me. How reliably does this problem reoccur for you? Every time?

I have DMA enabled, it stays enabled after reboot (both on c687a53 and 4950dbb) and the transfers are happening in DMA mode (indicated by `feature 0x01`).

With c687a53:

<pre>
[4427701ms] DBG Ejection button pressed
[4427703ms] DBG Image file debian-12.9.0-i386-netinst.iso size 676331520
[4427769ms] DBG Image file debian-12.9.0-i386-netinst.iso is contiguous, sectors 6557056 to 7878015
[4427770ms] Loading image "debian-12.9.0-i386-netinst.iso"
[4427771ms] DBG Image file debian-12.9.0-i386-netinst.iso size 676331520
[4427838ms] DBG Image file debian-12.9.0-i386-netinst.iso is contiguous, sectors 6557056 to 7878015
[4427840ms] DBG Image file debian-12.9.0-i386-netinst.iso size 676331520
[4427907ms] DBG Image file debian-12.9.0-i386-netinst.iso is contiguous, sectors 6557056 to 7878015
[4427907ms] -- Device loading media: "debian-12.9.0-i386-netinst.iso"
[4443096ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[4443097ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[4443099ms] DBG -- Reporting UNIT_ATTENTION condition after reset/medium change (ASC:0x3A00)
[4443100ms] DBG -- Command complete
[4443106ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x01, sector_count 0x01, lba 0xFF 0xFF 0x01)
[4443107ms] DBG -- ATAPI command: ATAPI_CMD_REQUEST_SENSE 0x03 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[4443108ms] DBG ---- ATAPI send 1x14 bytes: 0xF0 0x00 0x06 0x00 0x00 0x00 0x00 0x0B 0x00 0x00 0x00 0x00 0x3A 0x00 
[4443109ms] DBG -- ATAPI success
[4443109ms] DBG -- Command IDE_CMD_PACKET completed with error status 0x64
[4443113ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[4443114ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[4443115ms] DBG -- ATAPI success
[4443116ms] DBG -- Command complete
[4443116ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[4443117ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[4443119ms] DBG -- ATAPI success
[4443119ms] DBG -- Command complete
</pre>

[image]

[image]

morio, 2025-01-21:
I installed the VIA 4 in 1 driver so I  now have the VATAPI.VXD installed but the error is still occurring. Seeing the same issue in Bus Hound.

[zululog-4in1.txt](https://github.com/user-attachments/files/18495298/zululog-4in1.txt)

[image]

Log entry for the bus hound capture
```
[2712608ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[2712609ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[2712610ms] DBG -- ATAPI success
[2712610ms] DBG -- Command complete
[2714163ms] DBG Ejection button pressed
[2714166ms] Device ejecting media: "Quake2"
[2716608ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[2716609ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[2716638ms] DBG Image file CD1 Windows98IsDumb.iso size 3276800
[2716639ms] DBG Image file CD1 Windows98IsDumb.iso is contiguous, sectors 12599296 to 12605695
[2716640ms] Loading image "CD1 Windows98IsDumb.iso"
[2716647ms] DBG Image file CD1 Windows98IsDumb.iso size 3276800
[2716647ms] DBG Image file CD1 Windows98IsDumb.iso is contiguous, sectors 12599296 to 12605695
[2716664ms] DBG Image file CD1 Windows98IsDumb.iso size 3276800
[2716664ms] DBG Image file CD1 Windows98IsDumb.iso is contiguous, sectors 12599296 to 12605695
[2716665ms] -- Device loading media: "CD1 Windows98IsDumb.iso"
[2716673ms] DBG -- ATAPI error: 0x02 0x3A00 (NOT_READY)
[2716673ms] DBG -- Command complete
[2716678ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x01, sector_count 0x01, lba 0xFF 0xFF 0x01)
[2716679ms] DBG -- ATAPI command: ATAPI_CMD_REQUEST_SENSE 0x03 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[2716680ms] DBG ---- ATAPI send 1x14 bytes: 0xF0 0x00 0x02 0x00 0x00 0x00 0x00 0x0B 0x00 0x00 0x00 0x00 0x3A 0x00 
[2716681ms] DBG -- ATAPI success
[2716682ms] DBG -- Command IDE_CMD_PACKET completed with error status 0x24
[2716693ms] DBG -- IDE regs: STATUS:0x50 CMD:0xA0 DEV:0xA0 DEVCTRL:0x08 ERROR:0x00 FEATURE:0x01 LBAL:0x01 LBAM:0xFE LBAH:0xFF
[2718608ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[2718609ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[2718610ms] DBG -- ATAPI success
[2718611ms] DBG -- Command complete
[2718622ms] DBG -- IDE regs: STATUS:0x50 CMD:0xA0 DEV:0xA0 DEVCTRL:0x08 ERROR:0x00 FEATURE:0x00 LBAL:0x01 LBAM:0xFE LBAH:0xFF
[2722608ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[2722609ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[2722610ms] DBG -- ATAPI success
[2722611ms] DBG -- Command complete
```

dzlia, 2025-01-21:
In my case the driver setup is similar to what @morio showed - Microsoft driver for the secondary IDE controller, VIA driver for the VIA Bus Master PIC IDE Controller.

PetteriAimonen, 2025-01-22:
@morio Would it be possible to get a logic analyzer capture of this?

I'm a bit lost on what could be the cause.
The UDMA checksum is calculated on RP2040 and compared to the checksum sent by the host.
So the extra four bytes cannot be part of the same UDMA burst, otherwise there would be a checksum error.

It appears the four bytes are always repeated 16-bit value.
This makes it sound like a glitch in IORDY signal.
Then D0-D15 would be just a random value, which would be sampled twice by the host (falling and rising edge).
But that wouldn't explain why there is no checksum error.

I think there may be cases where FPGA leaves DMARQ active if the command is terminated with unfinished transfer.
For some reason my initial fix for that caused other DMA problems for you.

dzlia, 2025-01-30:
Some magic has happened - now I'm able to to switch between images in ZuluIDE working in the DMA mode.

Circumstances around it: my system got annoyingly unstable - random lockups, and regular lockups after 10-30 minutes of playing FIFI World Cup 98 with Voodoo 2. At the same time I've got some new memory sticks, some of which were broken. So, I started checking them in memtest86+ loaded from a floppy, and lockups were even there, so it was not a problem with OS/drivers. I deleted all removable hardware (including ZuluIDE and HDD) except floppy, still with no success. Then I switched to a PCI video card, since the mobo forces visiting BIOS setup in on changed hardware, together with installing a PCI video card I switched off APM (Automatic Power Management) in BIOS. After that memtest86+ started working fine on different memory modules (expect ones that I consider broken or incompatible with the mobo). Then I returned devices one by one, ZuluIDE first, and no any problem with stability. Later on first boot into Win98 I went to power management and set it up as turned on all the time.

Now no lockups and also in DMA with latest firmware I'm able to switch between images just perfectly. I associate this to disabling APM, but maybe something else (like mobo's ESCD updated or similar low-level stuff hidden from the user) is involved.

Maybe this story would allow @morio , who has access to the same motherboard, to reproduce scenario and this can help in getting to the root of the problem. At the same time, since I've changes many things in short period of time, all this could be not reflecting the reality and glitches come again after time.

aperezbios, 2025-01-30:
@dzlia  thank you for the follow-up. Yes, bad memory can cause all sorts of problems, including false positives such as the ones it would seem you've encountered. I'm going to close this issue for now, as it doesn't seem to be something we're going to be able to replicate at this time.

dzlia, 2025-01-30:
It is definitely not caused by bad memory, because I saw freezes with different memory modules (I tested overall 23 of them in various combinations), and, moreover, this doesn't explain why extra data tag appears on eject, while data copying works fine.

aperezbios, 2025-01-30:
@dzlia apologies then, I misunderstood. If you encounter these issues again, please feel free to re-open this issue.

PetteriAimonen, 2025-01-31:
If I understood correctly, the issue also occurred for @morio, so it is not likely to be just bad memory. APM could be a possible reason, as the motherboard I use for testing is ACPI-based.

dzlia, 2025-01-31:
@aperezbios this bug need to be reopened. Today I booted the computer, ZuluIDE was in DMA, and nothing has been changed with it. The eject problem re-appeared - it just doesn't work right as in the description, and I have made a screenshot of a broken file system content displayed (sometimes, not always) when I enter the disc after ejection.
@PetteriAimonen , according to this, maybe APM is not related, and it was just a coincidence, while this bug is an intermittent issue, or something else temporarily "cured" the problem last time.

[image]

dzlia, 2025-02-19:
@aperezbios please could you reopen the bug? I don't have permissions to do this.

@PetteriAimonen  I can add to it that I have replaced the motherboard to ASUS P3V133, based on VIA VT82C693A/VT82C596B, and still similar problems. Sometimes switching between images works fine in DMA, sometimes Windows 98 doesn't report images switched, sometimes garbage is displayed in the root directory. Note that it is an earlier VIA chipset comparing to the one I used when raised the bug.

aperezbios, 2025-02-19:
@dzlia reopened, however I'm a bit mystified as to why you would choose to replace your motherboard with another VIA-based motherboard when we are confident that the VIA chipset is the source of the issue.. This simply invites further difficulty, and seems like a very odd decision. Was there a reason for this that's not immediately obvious?

dzlia, 2025-02-20:
@aperezbios , my setup is not ZuluIDE-centric. I have got now a Slot 1 PIII CPU ready for use, and it is for FSB 133 MHz. I don't see any real alternative to a VIA-based motherboards, because the only alternative is BX, for which FSB 133 MHz is stressful overclocking for PCI/AGP cards.

Moreover, what I'm sure about is that almost every real CD-ROM drive is able to work fine with most VIA chipsets in the DMA mode. Otherwise this would become apparent decades ago. I don't know if it is a different in timings between eject/insert of CD disks between a real hardware (disk replace is slow) and ZuluIDE (disk replace is instant), or mismatch in implementation of DMA or any other thing, but ZuluIDE, which tends to emulate a real CD-ROM driver, fails to do so, where regular CD-ROM drives just work.

In any case, even if I switched today to, say, a i440BX + PIII-500 with 100 MHz FSB, then sooner or later I would anyway use a VIA-base motherboard, because it is more than half of my current assortment of them.
I hope that you will manage to make compatibility of ZuluIDE closer to real CD-ROM drives, and choosing a VIA-based mobo won't cause questions why it was chosen.

PetteriAimonen, 2025-02-21:
I agree that this is an actual incompatibility with ZuluIDE, but unfortunately I haven't been able to figure out a method how to get it resolved.

@dzlia By any chance, have you tried whether disabling "Auto insert notification" in the drive settings make a difference? With that disabled, you will have to press F5 for Windows to detect the disc change, but it doesn't poll the drive continuously, which might avoid the problem.

dzlia, 2025-02-23:
@PetteriAimonen I have disable auto insertion notification and enabled DMA. I'll update you about the results. However, due to problems with playing tracks for the game that I play (reported in #179) I currently use primarily a virtual drive. Will update what I observe in this setup anyway.

PetteriAimonen, 2025-02-24:
@dzlia Thanks. The ATA standard has a lot of variability between chipsets and driver implementations, so unfortunately there still remains some unknown corner cases.

PetteriAimonen, 2025-02-24:
Yeah, it may be that we will be able to arrange something with morio as time permits, he can capture logic analyzer traces without shipping the board. I may do another attempt with the setup I have at some point - previously the DMA issue was easy to reproduce but for some reason I'm not hitting this eject bug. We have to prioritize various development tasks so it may take some time before we get to this.

sashahart, 2025-04-09:
I tried to repro this with a more recent firmware on an Abit VH6T (which should have a VIA VT82C686B southbridge). Using Windows 98 SE, the ZuluIDE has the DMA box checked. If I'm at "My Computer" and repeatedly right click on the drive and choose "Eject" it switches quite smoothly.

(The following is not a formal bug report) I did see some quite flaky behavior with Windows 98 SE some months ago, on a different machine, and I have in the past had issues with the little eject button, so I would not say ZuluIDE has always been inherently trouble-free, or anything like that. But in this case the feature seems to be working well for me on this particular motherboard, as of this writing.

dzlia, 2025-04-24:
@PetteriAimonen I installed the latest firmware, and in the DMA mode on the mobo P3V133 switching between CD images works now. I will keep testing to make sure it is not working at random (which happened before). However, I recall one of the changes in firmware related to adding delay after ejecting the CD image, and maybe this helped (as I suspected as one of the reasons of failure earlier in this thread). Maybe @morio can check this using the IBM VIA Apollo 133A mobo that he has and which confirmed that the bug is reproducible only only for me.

I'll report after few days or weeks of testing, so please don't consider this post as a signal that the bug is really fixed.

dzlia, 2025-04-24:
Well, it doesn't work right. Even though each time on every eject the image is eventually reloaded, it can work in three different variations:
1) next image is "picked" by Windows 98 Explorer right after eject (or <1s after) - it is quite rare in my case.
2) instead of next image there is an icon of a drive without an image. The "My Computer" view of Explorer never refreshes to the next image, but "open" or double-click on the drive show content of the next image
3)
3a) _Auto Insert Notification is off._ The old image remains in "My Computer". On double click old autorun is attempted to be run, the context menu shows old autorun options. But right-click -> Open shows new content. And even after entering into the drive root "My computer" doesn't refresh the drive icon/label and its context menu. Only explicit refresh updates the state of the drive item.
3b) _Auto Insert Notification is on._ Same as above in many cases, but also very often there is a blue screen with error telling that the CD with the serial number <serial_number_of_old_image> needs to be inserted. I assume it is because it tries to auto-run the CD, and it detects a wrong CD is inserted and considers it as a error.

So, Windows 98 Explorer, "My Computer" page after ejection still often thinks that the old CD-ROM is still inserted in the drive. I cannot recall anything like this with a hardware CD-ROM drive back in a day when I used them. It looks to be a problem with how Zulu handles ejection and reports it to Windows 98.

aperezbios, 2025-07-06:
@dzlia thanks for the report. For now, please revert to the previous release that works for your environment, and we will investigate as time allows.