ZuluIDE/ZuluIDE-firmware #169: Make blacklisting of unused images / whitelisting of used images.

dzlia, 2025-01-17:
At the moment all files, including images in sub-directories are subject to be included into eject/remount loop. If there are many images on an SD card then it is annoying to switch between all them one by one. It could be useful if configuration allows the user to include only images/directories that are used currently and/or exclude images/directories that are not used. In the latter case currently unused images can be moved to a single directory which is then blacklisted.

dzlia, 2025-01-17:
as a workaround, I currently use a directory zulu_unused for images on the SD card that I don't use. It starts with zulu and therefore ignored. But this looks a bit hacky.

aperezbios, 2025-01-23:
@dzlia this is actually currently possible, but not properly documented. The ZuluIDE firmware already any ignores directory or file that begins with a _non-alphanumeric_ character, so if you prefix a directory name with a dot or #, it will never be enumerated by the firmware.

closed 2025-01-23 by aperezbios

dzlia, 2025-01-23:
Sounds good enough. Since I still don't see this on the home page, could documentation be updated, please?

aperezbios, 2025-01-23:
@dzlia I was in the middle of updating them when you wrote your response. This is now documented in the README and on the wiki.