ZuluIDE/ZuluIDE-firmware #170: Playing Audio CD is broken.

dzlia, 2025-01-20:
With the most recent firmware (16-Jan-2025) I have broken play Audio CD functionality, even in PIO mode. Without anything in logs, playing tracks with Windows 98 CD player and from games don't work anymore (while I see the track lists in the CD player), and I've got three complete system lockups, all times with ZuluIDE LED lighting constantly.

Attaching a log of the full working cycle from turning on the computer till one of the lockups that has happened with an attempt to switch from non-playing track 02 to the track 05.

It behaved in the same way for me for three different cue/bin images, so likely it is not an image-specific problem.

[zululast.txt](https://github.com/user-attachments/files/18481900/zululast.txt)
[zululog.txt](https://github.com/user-attachments/files/18481903/zululog.txt)

zuluide.ini:
```
[IDE]
debug = 1
enable_usb_mass_storage = 1
```

dzlia, 2025-01-20:
I switched to the firmware v2025.01.06. After loading in PIO, with the same image (GROMADA), the CD player still didn't want to start playing the tracks. Then I switched to a different image (HEROES2) using software eject, and CD audio started playing, and it was playing for every next image, including GROMADA, which didn't want to play right after system startup. What I see is that in v2025.01.06 it works better for me, but still with glitches. v2025.01.16 seems to work worse. However, it could be that it is an intermittent problem.

morio, 2025-01-21:
It looks like there were two errors, one stopped the first loaded CD from playing audio, the other disabled audio playback all together.
Checkout this build https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/12877926258

The DMA issue is separate to this.

dzlia, 2025-01-21:
@morio with the proposed firmware I'm able to play CD audio from the last used image, so it looks like the defect is fixed.

aperezbios, 2025-01-21:
@dzlia we've now released https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2025.01.21 which incorporates this fix.

closed 2025-01-21 by aperezbios

dzlia, 2025-01-21:
Yes, this firmware works for me, too.