ZuluIDE/ZuluIDE-firmware #173: Audio playback is not stopped by accessing data on mixed-mode CD.

dzlia, 2025-01-23:
I noticed that if there is a mixed-mode (data + audio) CD and CD Audio is being played, then playback doesn't stop even if I access CD data (i.e. read files from it). This is not how CD-ROM devices works and it is annoying, because the way to stop playback is to eject the CD, because for a CD audio player in Windows 98 reports that the CD is already busy and doesn't give control over it.

The situation is that I run a game which started the playing CD audio and then I had to close it using the hotkey Ctrl+F4, which is an abnormal exit for this game. CD audio continued playing with the above described effect.

Observed on:
VIA south bridge, Windows 98, PIO, Carmageddon BIN/CUE image, DOS version run under Windows 98.

If there are reasons to have this behaviour then at least it can be made configurable, so that users can decide whether or not simultaneous playback and accessing data is good (for convenience) or bad (because it is not how CD-ROM work).

P.S. Ironically, in Carmageddon's README there is a note that CD-Audio will work only with complete installation because in smaller installations audio playback will be start-stopped all the time because of frequent fetching game data from the CD.

morio, 2025-02-28:
@dzlia 
This should fix the issue of CD reads not stopping audio playback https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/13595957460

It won't stop audio if you ALT-F4 out of Carmageddon because Win9x doesn't emit a SCSI command that would stop the audio playback. I also noticed even if you ALT-F4 out of Carmageddon, the executable is still listed in the task manager which is why the CD-Audio player reports the device being busy. Terminating the task frees the CD-ROM drive. 

Hope that helps

dzlia, 2025-03-01:
@morio, I briefly checked it with a CD audio player. Before the fix accessing the CD didn't have any effect on playback. With the fix the CD audio player in Windows 98 stops playing the current track and immediately starts playing the next one (until all tracks are over). I don't have a hardware CD-ROM connected anywhere to check how it behaves, but if you do then could you please check if works in the same way for ZuluIDE and a CD-ROM?

morio, 2025-03-12:
@dzlia I don't have an Mixed Data/Audio CD to test with, but as far as I can tell this behavior is what Windows CD Player does normally.
Below is a log capture
```
[80673ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x15 0xAD 0x00 0x00 0x01 0x00 0x00 0x00
[80674ms] DBG -- Read 1 sectors starting at 5549
[80675ms] DBG ------ CD-ROM Stop Audio request
[80675ms] DBG ---- Read CD: 1 sectors starting at 5549, track number 1, sector size 2352, main channel 0x10, sub channel 0x00, data offset in file 13051248
...
[80936ms] DBG -- ATAPI command: ATAPI_CMD_READ_SUB_CHANNEL 0x42 0x02 0x40 0x01 0x00 0x00 0x00 0x00 0x10 0x00 0x00 0x00
[80937ms] DBG ------ Get audio playback position: status 19 lba 0
[80937ms] DBG ------ ABS M 0x00 S 0x02 F 0x00
[80938ms] DBG ------ REL M 0x00 S 0x00 F 0x00
[80938ms] DBG ---- ATAPI send 1x16 bytes: 0x00 0x13 0x00 0x0C 0x01 0x14 0x01 0x01 0x00 0x00 0x02 0x00 0x00 0x00 0x00 0x00
[80940ms] DBG -- ATAPI success
[80940ms] DBG -- Command complete
[80940ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
...
[81027ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00
[81029ms] DBG -- ATAPI success
[81029ms] DBG -- Command complete
[81029ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x08, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[81031ms] DBG -- ATAPI command: ATAPI_CMD_PLAY_AUDIO_MSF 0x47 0x00 0x00 0x26 0x12 0x06 0x47 0x36 0x2B 0x00 0x00 0x00
[81032ms] DBG ------ CD-ROM Play Audio request at 172206 for 151237 sectors
[81033ms] DBG ------ CD-ROM Stop Audio request
[81033ms] DBG ------ Play audio CD: 151237 sectors starting at 172206, track number 3, data offset in file 405026160
[81034ms] DBG ------ Audio seek required
[81041ms] DBG -- ATAPI success
``` 

I started playing track 2, opened up a folder on the CD (READ10) which stops play back and then about half a second later the host requested to play track 3 (PLAY_AUDIO_MSF). 

I'm not totally sure if the READ_SUB_CHANNEL command should be zeroed out after a stop (the SCSI 2 spec is not clear on this), so I modified the code to keep the last playback position and Windows CD Player still started playing the next track.

I'm about 80% sure this is the expected behavior.

morio, 2025-03-13:
Tested a physical mixed mode CD with a Drive and CD Player advanced the track just like with ZuluIDE.