ZuluIDE/ZuluIDE-firmware #174: ZuluIDE does not initialise itself after cold boot.

dzlia, 2025-01-25:
A couple of times (not always) I saw with 2025.01.21-release that it locks for some time POST and starts flashing with a couple of slow signals and fast signals in loop. Then the system starts, but the CDROM is not present. Attaching a log collected from the SD card when I disabled computer after these "SOS" signals.

It is definitely something new, I don't recall it to fail that frequently with older non-experimental firmware revisions that I used.

[zululog.txt](https://github.com/user-attachments/files/18544369/zululog.txt)

System setup is the same: VIA Apollo 133A, Secondary slave ZuluIDE (the only device on the channel).

aperezbios, 2025-01-25:
@dzlia does this happen 100% of the time, or only sometimes? Did it write a zuluerr.txt file by chance?

dzlia, 2025-01-25:
It has happened to me at least 3 times, but there are also successful boot attempts when the device works as usual. Hard to say how frequent the problem is, but it is definitely not a sudden glitch caused by cosmic rays.

JanSchr, 2025-03-13:
Same here, updated to firmware v2025.02.28 today. When directly booting to MS-DOS 6.22 the CDROM is not present (OAKCDROM.SYS detects no drive). 

Fun fact: When booting Windows 98 SE it's there, then restarting the PC (without turning it off) and booting MS-DOS 6.22 fixes this for MS-DOS 6.22, it's present then.  Once you turn of the PC again and then boot it again directly to MS-DOS 6.22 it's gone again.

aperezbios, 2025-03-13:
> Same here, updated to firmware v2025.02.28 today. When directly booting to MS-DOS 6.22 the CDROM is not detected (OAKCDROM.SYS detects no drive).
> 
> Fun fact: When booting Windows 98 SE it's there, then restarting the PC (without turning it off) and booting MS-DOS 6.22 fixes this for MS-DOS 6.22, it's present then. Once you turn of the PC again and then boot it again directly to MS-DOS 6.22 it's gone again.

@JanSchr please clarify whether or not it's only happening with the new firmware, or if you are not sure. It's not at all clear from your initial "same here" comment.

JanSchr, 2025-03-13:
Okay happening with new firmware, never seen this behavior before. Should also add that device is not recognized by bios anymore (also only on new firmware). Once windows 98 has been started and PC is restarted bios recognizes device again during boot.

aperezbios, 2025-03-13:
@JanSchr can you confirm that your ZuluIDE starts behaving as expected once you downgrade the firmware back to the previous release? You also didn't state which firmware revision you upgraded from, which is a piece of information we absolutely need, if you know what it was at prior to the upgrade.

dzlia, 2025-03-14:
@JanSchr , in my case it was 3-5 times at random moments, and the device just stopped at self initialisation, blinking SOS, and the cold reboot helped. Also I didn't see it for at least 1.5 months. In your case it looks to be a repetitive issue starting from some firmware release, with Windows 98 seeing the device, so self initialisation has likely been successful for ZuluIDE.
So maybe these problems have different cause.

JanSchr, 2025-03-14:
@dzlia Probably you're right. In my case it's not random at all but a 100% reproducible, also never observed any SOS blinking. Sorry for "hijacking" your issue!

dzlia, 2025-03-22:
@PetteriAimonen I saw it 3-4 times during the whole period. Next time it fails I will make a video recording of blinking and extract what I can. I'm not stuck to a single firmware, because I reported a series of bugs, and every verification makes me jumping to another (most of the time - development) releases of the firmware. But for sure I will provide the firmware version used if the crash happens next time.

I don't know what is USB serial port in terms of ZuluIDE. I have got a couple of USB->RS232 adapters, so if in this case I can attach them to ZuluIDE and read some diagnostic data, then in theory I can do this, but I need an instruction what exactly to do.

aperezbios, 2025-03-22:
@dzlia the serial port isn't a physical RS-232 port, it's a virtual "COM port" that will show up if you simply attach a USB-A/C cable to a modern computer.  It works via USB. There is no separate serial interface.

dzlia, 2025-03-22:
@aperezbios , ok, after connecting it to my Debian I see a new tty device: /dev/ttyACM0 . Still an instruction is needed how to use it to dump state in case of failure. Just because it is a physical protocol, without any logical protocol pre-defined on top of it.

Here is an attempt to read something from it:
```
# stty < /dev/ttyACM0
speed 9600 baud; line = 0;
-brkint -imaxbel
```

aperezbios, 2025-03-23:
If you have screen installed, you can run “screen /dev/ttyACM0” and attach to it. Control-A D disconnects the screen session.

dzlia, 2025-03-23:
@aperezbios , I've tried to connect but after initial fast LDE-flickering I see just a blank screen with no any reaction to my input (Ctrl+A+D finished the session). Will is report something upon connection in case of firmware failure?

dzlia, 2025-04-01:
It failed to start again. Fetching data from the serial port is slow as hell, but some data is coming in small chunks. It took more than 30 minutes to get first 3 seconds of device bootstrap.

Uploading now 1.5 minutes of video with LED blinking. The URL will die in two days, so please download it before it dies. I'll keep video locally for a while.
https://streamable.com/tt3kia

As far as I understand it blinks some data in Morse encoding and then same chunk of data becomes available on the terminal. And this is bloody slow.

I guess this is it:
```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.02.28-release Mar 19 2025 20:27:26
[11ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[12ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x2183434330C863E6
[99ms] FPGA initialization succeeded
[99ms] I2S audio to expansion header enabled
[100ms] -- System clock is set to 135428571Hz
[101ms] Hardware UI not found.
[101ms] I2C server not found
[110ms] SD card detected, FAT32 volume size: 59607 MB
[110ms] SD MID: 0xFE, OID: 0x34 0x32
[110ms] SD Name: SD
[110ms] SD Date: 7/2024
[111ms] SD Serial: 0x00000044
[123ms] Waiting for USB enumeration to enter Card Reader mode.
[1124ms] Selecting device type when loading first image.
[1197ms] Device is a CDROM drive
[1203ms] Device 0 configuration:
[1203ms] -- Max PIO mode: 3 (phy max 3)
[1203ms] -- Max UDMA mode: 0 (phy max 0)
[1204ms] -- Max blocksize: 4096 (phy max 4096)
[2224ms] Loading last used image: "quake2"
[2225ms] Loading image "quake2"
[2311ms] ---- Cue sheet cd.cue loaded with 11 tracks
[2320ms] Initialization complete!
[2321ms] --------------
[2321ms] CRASH!
[2321ms] Platform: ZuluIDE RP2040
[2321ms] FW Version: 2025.02.28-release Mar 19 2025 20:27:26
[2322ms] SP: 0x20040F88
[2322ms] PC: 0x2000FC50
[2322ms] LR: 0x2000FD1B
[2323ms] R0: 0x00000000
[2323ms] R1: 0x00000000
[2323ms] R2: 0x000000EB
[2323ms] R3: 0x0000009A
[2324ms] STACK 0x20040F88:    0x00000000 0x00000000 0x000000EB 0x0000009A
[2324ms] STACK 0x20040F98:    0x10025575 0x2000FD1B 0x2000FC50 0x21000000
[2325ms] STACK 0x20040FA8:    0x89FD4CA5 0xA65F4801 0xB172E502 0x12923A16
[2326ms] STACK 0x20040FB8:    0x2002E798 0x00000180 0xE000E100 0x10026C31
[2327ms] STACK 0x20040FC8:    0x00000178 0x10026C31 0x00000178 0x00000179
[2327ms] STACK 0x20040FD8:    0xE000ED00 0x2000FD1B 0x10026C31 0x2000F313
[2328ms] STACK 0x20040FE8:    0x10026C31 0x10026C55 0x10026C31 0x1002A25D
[2329ms] STACK 0x20040FF8:    0xD0000000 0x0000017F 0x02A13DD0 0xBA0F4F8E
[2330ms] Only core 0 may attempt 'platform_emergency_log_save()'
```

PetteriAimonen, 2025-04-02:
@dzlia Thanks, I have downloaded the video in case I need it, but the text output is more useful in any case.

The mystery of not getting zuluerr.txt is caused by this being core1 code that is crashing.
Apparently log saving doesn't happen at all in that case.

The bug itself appears to be a race condition between the two CPU cores when external UI control panel is not present.
Sometimes the second core would end up accessing a null pointer it before the first core stops it.

PetteriAimonen, 2025-04-02:
Hmm, now that I think about it again, maybe that crash & blinking is a false clue - core 1 crashes, but that is only visible because core0 is hanging somewhere. But at least one bug will be fixed, and hopefully we get something more revealing if the core0 hangs continue. I think watchdog should catch it.

But it could also be that the core1 crash handler is somehow disturbing core0 operation, because they both call into USB handlers.

PR #194 (2025-04-03): Fix null pointer dereference in core1 UI controller code
It seems sometimes core1 would have enough time to access uiStatusController which is null before the zuluide_setup() runs to stop it.

@hoffman373 Just a ping in case you want to check whether I have missed anything. It seems like a simple race condition between `zuluide_main_loop1()` accessing `uiStatusController` when it is null, vs. `zuluide_setup()` running `multicore_reset_core1()`.

Fixes #174
closed 2025-04-03 by aperezbios