ZuluIDE/ZuluIDE-firmware #178: CD audio cuts off with some AmigaOS software

Aardvark1982, 2025-02-02:
Some AmigaOS CD players and games cuts off the audio after few seconds. This doesn't occur when using real CD-ROM drive.
Here are logs of using different CD players. First log plays correctly, the second log cuts off within seconds.

[zululogCDAudioSuccess.txt](https://github.com/user-attachments/files/18630688/zululogCDAudioSuccess.txt)
[zululogCDAudioFail.txt](https://github.com/user-attachments/files/18630689/zululogCDAudioFail.txt)

The CD player that cuts off sends "ATAPI_CMD_PLAY_AUDIO_12", while the working CD player sends "ATAPI_CMD_PLAY_AUDIO_MSF".

I understand that ZuluIDE may not have been designed or tested for the Amiga, but still would be interesting to hear a theory for this behavior, or if there's any INI settings I could try.

Hardware: Amiga 1200 with 68060 accelerator. ZuluIDE is connected to A1200 IDE via buffered 4xIDE adapter alongside with Compact Flash hard disk.

aperezbios, 2025-02-02:
Thanks for the report @Aardvark1982, we will investigate, attempt to replicate, and report back in due time.

Aardvark1982, 2025-02-26:
Working great now, thanks!

morio, 2025-02-28:
Thanks for the follow up!