ZuluIDE/ZuluIDE-firmware #179: Wrong audio track offsets and playback on a mixed CD.

dzlia, 2025-02-20:
There is an image of mixed data+audio content. And I noticed, then it is used in ZuluIDE, that beginning of the next track is moved to the end of the previous track, so the border between tracks is shifted to ~4 seconds. When I mount the same image to an Alcohol 120% virtual drive, everything's fine.

In addition, when I play the track (track 5 in my case) using Win 98 CD Audio player, in the end of the track it quickly shifts from track 6 to track 9, while the track 6 is still played after the track 5. If needed, I can record a video of how this works in dynamics.
In addition, the overall duration of the whole CD and audio tracks is reported incorrectly.

Here is an image (this page is in russian, use translator to understand how to download the image, since it generates a dynamic download link): https://www.old-games.ru/game/download/get.php?fileid=26015&modal=1

Here are screenshots of the difference between how an image mounted in Alcohol 120% and Zulu IDE is reported.
[image]
[image]

dzlia, 2025-02-20:
From checking CUE, I have a guess that CUE with pre-gaps is not yet supported. This CUE was generated by ImgBurn, so it should be a valid one.

morio, 2025-02-25:
@dzlia could you try this build https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/13512136892

It should fix the issue with alcohol 120% and Windows CD Player.

I just pushed it end of day so I've only tested it with the CD you mentioned in this issue.

dzlia, 2025-02-25:
@morio , this firmware works for me, too, for the same CD.

morio, 2025-02-28:
Thanks for the update!
Code merged into main

closed 2025-02-28 by morio