ZuluIDE/ZuluIDE-firmware #184: Crash when trying to launch Warcraft III

TuscanRaider, 2025-03-01:
Launching freshly installed Warcraft III from iso will freeze / lock Windows 98 (4.10.2222).  The launch window will appear but freezes seconds after the launch

Pentium 233 MMX / 128 MB RAM / Asus P55T2P4 Rev 3

CD-ROM is detected as Mode 3

Zuluide.ini

[IDE]
Device = CDROM
init_with_last_used_image = 0
reinsert_media_on_sd_insert = 1

Zuluerr.txt

[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.02.28-release Feb 28 2025 20:46:02
[11ms] DIP switch settings: cablesel 0, drive_id 1 debug log 0
[12ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x21272E4330C863E6
[99ms] FPGA initialization succeeded
[99ms] I2S audio to expansion header enabled
[100ms] -- System clock is set to 135428571Hz
[101ms] Hardware UI not found.
[101ms] I2C server found
[118ms] SD card detected, FAT32 volume size: 30424 MB
[118ms] SD MID: 0x03, OID: 0x53 0x44
[118ms] SD Name: SK32G
[118ms] SD Date: 5/2024
[119ms] SD Serial: 0x8749C913
[133ms] Device is a CDROM drive
[133ms] Initialized platform with device control.
[135ms] Set SSID from INI file to saturn
[136ms] Set PASSWORD from INI file.
[136ms] Initialized platform controller with the status controller.
[139ms] Initialized platform controller with the display controller.
[139ms] Initialized platform controller with input receiver.
[152ms] Device 1 configuration:
[152ms] -- Max PIO mode: 3 (phy max 3)
[152ms] -- Max UDMA mode: 0 (phy max 0)
[152ms] -- Max blocksize: 4096 (phy max 4096)
[175ms] Loading first image 1WINXP3-SP3p.iso
[175ms] Loading image "1WINXP3-SP3p.iso"
[247ms] Initialization complete!
[270ms] I2C Client subscribed to updates.
[2000ms] FPGA license request code: 0xE6 0x63 0xC8 0x30 0x43 0x2E 0x27 0x21 0xFF 0xBF 0xFF 0xFF 0x98 0x28 0x4B 0x0A 0xBA 
[2000ms] FPGA license accepted with status 0x81
[7827ms] Client IP address is: 192.168.5.166
[21585ms] -- Set PIO transfer mode 3
[189450ms] Device ejecting media: "1WINXP3-SP3p.iso"
[189760ms] Loading image "BLADE_1.ISO"
[189884ms] -- Device loading media: "BLADE_1.ISO"
[214757ms] Device ejecting media: "BLADE_1.ISO"
[215118ms] Loading image "BLADE_2.ISO"
[215246ms] -- Device loading media: "BLADE_2.ISO"
[217927ms] Device ejecting media: "BLADE_2.ISO"
[218218ms] Loading image "BLADE_3.ISO"
[218341ms] -- Device loading media: "BLADE_3.ISO"
[220126ms] Device ejecting media: "BLADE_3.ISO"
[220393ms] Loading image "BLADE_4.ISO"
[220510ms] -- Device loading media: "BLADE_4.ISO"
[224352ms] Device ejecting media: "BLADE_4.ISO"
[224433ms] Loading image "CC_GDI.ISO"
[224535ms] -- Device loading media: "CC_GDI.ISO"
[227031ms] Device ejecting media: "CC_GDI.ISO"
[227255ms] Loading image "CC_NOD.ISO"
[227352ms] -- Device loading media: "CC_NOD.ISO"
[229185ms] Device ejecting media: "CC_NOD.ISO"
[229453ms] Loading image "GRIM_DISC_A.ISO"
[229590ms] -- Device loading media: "GRIM_DISC_A.ISO"
[236981ms] Device ejecting media: "GRIM_DISC_A.ISO"
[237280ms] Loading image "GRIM_DISC_B.ISO"
[237417ms] -- Device loading media: "GRIM_DISC_B.ISO"
[252287ms] Device ejecting media: "GRIM_DISC_B.ISO"
[252576ms] Loading image "TheFrozenThrone.iso"
[252680ms] -- Device loading media: "TheFrozenThrone.iso"
[258160ms] Device ejecting media: "TheFrozenThrone.iso"
[258422ms] Loading image "Warcraft III.iso"
[258559ms] -- Device loading media: "Warcraft III.iso"
[318949ms] --------------
[318949ms] CRASH!
[318949ms] Platform: ZuluIDE RP2040
[318949ms] FW Version: 2025.02.28-release Feb 28 2025 20:46:02
[318950ms] SP: 0x2003FE98
[318950ms] PC: 0x1002A564
[318950ms] LR: 0x1002A637
[318951ms] R0: 0x00000001
[318951ms] R1: 0xD0000148
[318951ms] R2: 0x00000000
[318951ms] R3: 0x1002A631
[318952ms] STACK 0x2003FE98:    0x00000001 0xD0000148 0x00000000 0x1002A631
[318952ms] STACK 0x2003FEA8:    0x200002C5 0x1002A637 0x1002A564 0x21000000
[318953ms] STACK 0x2003FEB8:    0x1003E34C 0x1003D97C 0x1003D97C 0x2003FEDF
[318954ms] STACK 0x2003FEC8:    0x2003FF90 0x200019FD 0x00000000 0x2003FF90
[318955ms] STACK 0x2003FED8:    0x2001342C 0x8D00FFFE 0x20013358 0x200073D3
[318955ms] STACK 0x2003FEE8:    0x00000000 0x2001341C 0x00000100 0x20000E0B
[318956ms] STACK 0x2003FEF8:    0x00000000 0x00000000 0x00000000 0x20001849
[318957ms] STACK 0x2003FF08:    0x00000000 0x20013358 0x2003FF90 0x2003195A

TuscanRaider, 2025-03-01:
[zululog.txt](https://github.com/user-attachments/files/19036855/zululog.txt)

TuscanRaider, 2025-03-01:
[zuluerr.txt](https://github.com/user-attachments/files/19036897/zuluerr.txt)

PetteriAimonen, 2025-03-03:
@TuscanRaider Thanks for the logs.

There seems to be two kinds of problems happening. `IDEATAPIDevice::atapi_send_wait_finish() data write timeout` during `ATAPI_CMD_REQUEST_SENSE` which would cause 10 second pauses but seems to recover by itself. And then the crash from assertion error after reception of the ATAPI packet command fails for some reason.

I think I'll add some recovery logic for the latter, but the timeouts indicate that it is more of a symptom rather than cause.

I'll try to reproduce this problem tomorrow on my hardware, as that would simplify debugging a lot.

Is your .iso the one from [archive.org](https://archive.org/details/warcraft-iii_202406)?

aperezbios, 2025-03-04:
@TuscanRaider  ping.

TuscanRaider, 2025-03-04:
Actually, this is my image from the original disk, created from a bluray drive.

I am getting the image from archive.org as we speak and will test that.

PetteriAimonen, 2025-03-05:
@TuscanRaider Thanks for the updates. I'm glad to hear you got it working :)

@aperezbios Do you feel that we should try to present better diagnostics or otherwise improve behavior in this case, or just close this as resolved?

aperezbios, 2025-03-05:
@PetteriAimonen at a bare minimum, I think we need to be able to handle the case without a hard crash on the firmware side, if possible. Maybe we could also log a message that SecuROM images aren't yet fully supported

TuscanRaider, 2025-03-07:
yes will try

PetteriAimonen, 2025-03-11:
Thanks for the log.
Something weird is definitely happening there, but I don't currently know how we could figure out what.

The IDE communication breaks down, but it's unclear whether the host side controller gets confused or if it triggers some weird bug in ZuluIDE. In either case, it would be great if we could avoid that happening, but I think in this particular case it is not worth the effort.