ZuluIDE/ZuluIDE-firmware #185: Crash with OS lockup

dzlia, 2025-03-05:
I played a game which uses CD audio. Suddenly I heard scratchy sound (like DJs' manually rotating vynil), then OS lockup and ZuluIDE started flashing S.O.S. Attaching error and log files.

[zuluerr.txt](https://github.com/user-attachments/files/19098428/zuluerr.txt)
[zululog.txt](https://github.com/user-attachments/files/19098427/zululog.txt)

aperezbios, 2025-03-05:
@PetteriAimonen please investigate as time allows.

PetteriAimonen, 2025-03-06:
This appears very similar to issue #184. Both cases involve an assert failure at ide_phy_can_read_block() at the start of cmd_packet() and the execution of audio playback commands. I have kind of an idea how to avoid the crash, but I'll try digging deeper to find the actual cause.

Is this something that happens every time, or did it happen only one time?
Just asking in case you have time to test once I have a probable fix for this.

dzlia, 2025-03-06:
@PetteriAimonen , it happened only once in a playing session lasting for 1.5-2 hours. I don't have any problems with testing it - as you can see it was already a dev release of the firmware, so I'm a kind of hired tester.

PetteriAimonen, 2025-03-07:
Yeah, it's just hard to know whether things have been fixed with random bugs.

In any case, this version (PR #494) tries to avoid the crash, hopefully being able to recover gracefully (but probably with some temporary pause / glitch):
https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/13717816820

dzlia, 2025-03-07:
@PetteriAimonen , I'll try it. However, in case of glitch/recovery - should there be any record written to zululog.txt or zuluerr.txt? How would I know that there was glitch and subsequent recovery?

dzlia, 2025-03-07:
Ok, from the PR I see that first debug must be enabled and then there will be just regular log messages among others, right?

PetteriAimonen, 2025-03-07:
Yeah, and there will probably be a lot of debug log in normal use.

I would just use it without debug logging and see if you get any more
crashes. If not, either it helped or you are lucky :)

dzlia, 2025-03-07:
@PetteriAimonen I have installed this firmware to my ZuluIDE device. If you don't hear from me here then it should mean that I don't see new crashes with it.

dzlia, 2025-03-14:
@PetteriAimonen, for now no crashes, but also it's worth keeping in mind that it has happened to me only once before the fix.

dzlia, 2025-03-14:
Hi @PetteriAimonen, please check #187 , just in case it is related to the crash reported here.

aperezbios, 2025-05-05:
@dzlia I'm going to close this as we haven't heard anything further in over 50 days, and I believe there are other open issues that encapsulate the remaining problems you've encountered.

closed 2025-05-05 by aperezbios

dzlia, 2025-05-05:
yes, no crashes so far, and all other glitches I see are reported.