ZuluIDE/ZuluIDE-firmware #187: CD audio stops working until power off.

dzlia, 2025-03-14:
In Windows 98 SE, I played with Control Panel -> Multimedia -> CD Audio, volume slider. When I moved it back and force while a mixed CD (Heroes of Might and Magic II) was playing, the sound disappeared. CD player showed activity, but there was no sound from CD. MIDI worked well at the same time. I rebooted the OS using restart feature (without cold reboot) - still no sound from CD. Then I powered off the computer, turned it on again - sound appeared, until I moved that slider again. Since it is cured by cold reboot, I assume it is a problem in ZuluIDE, not in Windows 98 SE.
[image]

Alcohol 120%'s CD didn't react on this slider. And I'm not able to check how a physical CD-ROM drive reacts on this slider.

There is nothing interesting in zululog.txt, but I can record it in debug mode if there are problems with reproducing it.
[zululog.txt](https://github.com/user-attachments/files/19255302/zululog.txt)

There is also zuluerr.txt, but it is from an earlier crash. For some reason it is not deleted automatically, as it is done for zululog.txt. Maybe, to make it available when ZuluIDE's SD card is accessed via USB, to copy this file.

PetteriAimonen, 2025-03-15:
Debug log might shed more light.

But most useful would be for me to be able to reproduce the problem. Is the game image something that I can download? Or even better, does the same thing occur if you play a pure music CD and adjust the slider?

dzlia, 2025-03-15:
@PetteriAimonen , I reproduce it easily on many mixed mode CDs (don't have any Audio CD images loaded). Literally: Heroes of Might and Magic II, WinQuake, Carmageddon. For instance, you can check this CD (Carmageddon): https://www.old-games.ru/game/download/get.php?fileid=23860&modal=1
You would probably need to translate the page from russian into English to understand what is the download URL, which will be dynamically generated.

If there are problems with reproducing it, I'll collect debug logs.

morio, 2025-03-19:
@dzlia try this build https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/13956043720

Saw this issue in Linux awhile ago, thought it was a kernel bug. Turns out Mode Sense for CD-ROM audio doesn't return the values set by MODE SELECT, but returns their mask values.

closed 2025-03-19 by morio

dzlia, 2025-03-22:
@morio ok, will try

dzlia, 2025-03-22:
@morio , it works and CD audio volume is now adjustable with this slider (unlike how it works for Alcohol 120%, where it just doesn't do anything). Since I attached ZuluIDE to a separate channel of an external mixer, I can confirm that volume is adjusted directly inside ZuluIDE.