ZuluIDE/ZuluIDE-firmware #190: Award BIOS fails to boot from ZuluIDE, succeeds after soft reboot from Linux

joshlitherland, 2025-03-21:
Firmware version 2025.02.28-release.

Scenario 1:
Attempting to cold boot from an ISO image with ISOLINUX bootloader on a Pentium III system with Award Modular BIOS.  Read commands fail with watchdog timeouts.  See attached zululog-fail.txt.

Scenario 2:
Start with no SD card in ZuluIDE.  Allow system to boot to Linux on hard drive.  Insert SD card with same ISO image as in first scenario, then soft-reboot.  Now, ZuluIDE boots successfully from ISO image, reads proceed quickly and without error.  See attached zululog-succeed.txt. 

Would appreciate any troubleshooting advice you can provide.  Thank you!

[zululog-fail.txt](https://github.com/user-attachments/files/19387585/zululog-fail.txt)
[zululog-succeed.txt](https://github.com/user-attachments/files/19387584/zululog-succeed.txt)

[zuluide_ini.txt](https://github.com/user-attachments/files/19387620/zuluide_ini.txt)

aperezbios, 2025-03-21:
@joshlitherland thanks for the report. We need a little bit more information about your actual hardware before we can attempt to replicate it. Can you please provide the make, model, and revision of your motherboard? This information will allow us to determine the underlying information we need.

joshlitherland, 2025-03-21:
Sure!  I don't see a revision code per se, but the board is an Abit AB-BX6.  Some information about this board can be found here: 

https://soggi.org/motherboards/abit/BX6.htm

sashahart, 2025-04-09:
I attempted to repro this on a Abit VH6T (Phoenix BIOS, which I'm aware is not the same as OP's Award BIOS). I looked at the fail log and saw a filename of an Arch image, which hopefully I located at https://mirror.math.princeton.edu/pub/archlinux32/archisos/archlinux32-2024.07.10-i686.iso. md5sum d66328286370366cae23519169b1c728. I "inserted" this CD and rebooted. In one case, with ctrl-alt-del; in the other case, from a cold power off to hitting the power switch with the CD in. In both cases, straight into Grub, where I could run things like the Hardware Detection Tool. Arch booted (though it had some unrelated problem when it tried to switch out of the console to some kind of GUI mode).

I fully realize how unhelpful it is to say (in this case literally) "works on my machine," but at least it does rule out general variables like a Pentium III booting this Arch image from a ZuluIDE on cold boot.