ZuluIDE/ZuluIDE-firmware #193: Install process for Fallout 2 crashing Windows 98 PC consistently

timesscar, 2025-04-01:
During the install process for Fallout 2 using various archive images, the PC consistently crashes at the same point in the install process.  This symptom only occurs while using the zuluide.  The zuluide log is unremarkable.  ```[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.03.25-release Mar 25 2025 19:48:10
[11ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[12ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x22BD3E0F184064E6
[99ms] FPGA initialization succeeded
[99ms] I2S audio to expansion header enabled
[100ms] -- System clock is set to 135428571Hz
[101ms] Hardware UI not found.
[101ms] I2C server not found
[117ms] SD card detected, FAT32 volume size: 30416 MB
[118ms] SD MID: 0x03, OID: 0x53 0x44
[118ms] SD Name: SK32G
[118ms] SD Date: 4/2024
[119ms] SD Serial: 0x869CC68C
[151ms] Selecting device type when loading first image.
[249ms] Device is a CDROM drive
[281ms] Device 0 configuration:
[281ms] -- Max PIO mode: 3 (phy max 3)
[281ms] -- Max UDMA mode: 0 (phy max 0)
[281ms] -- Max blocksize: 4096 (phy max 4096)
[1362ms] Loading last used image: "Fallout 2 (USA)"
[1362ms] Loading image "Fallout 2 (USA)"
[1506ms] ---- Cue sheet Fallout 2 (USA).cue loaded with 1 tracks
[1521ms] Initialization complete!
[2000ms] FPGA license request code: 0xE6 0x64 0x40 0x18 0x0F 0x3E 0xBD 0x22 0x7F 0xFF 0xFF 0xFD 0xCD 0x37 0x32 0x66 0x81 
[2000ms] FPGA license accepted with status 0x80
```

Additionally, the zuluide still blinks when the SD card is removed, which leads me to believe the device firmware is not crashed.

![Image](https://github.com/user-attachments/assets/050712f2-2b1e-451a-b75f-810c62bb425a)

aperezbios, 2025-04-01:
@timesscar  please also describe the model of the PC/motherboard you're using, as this will tell us what sort of chipset the machine has.

timesscar, 2025-04-01:
> [@timesscar](https://github.com/timesscar) please also describe the model of the PC/motherboard you're using, as this will tell us what sort of chipset the machine has.

It is an ITX Llama board.  I will follow up with debug logs this evening.

timesscar, 2025-04-02:
`[3018262ms] Client requested the current image be set to:Fallout 2 %USAS
[3026082ms] SD card removed, trying to reinit
quested the current image be set to:Fallout 2 %USAS
[3293290ms] Device ejecting media, image already cleared
[3295333ms] Loading image "Fallout 2 (USA)"
[3295344ms] -- Device loading media: "Fallout 2 (USA)"
[3295396ms] ---- Cue sheet Fallout 2 (USA).cue loaded with 1 tracks
[3629370ms] --------------
[3629370ms] WATCHDOG TIMEOUT, attempting bus reset
[3629371ms] GPIO states: out 0x00FE0803 oe 0x0207007F in 0x33FF09FB
[3629371ms] STACK 0x2003FB00:    0x1003B235 0x33FF09FB 0x20011974 0x1002ABEF
[3629372ms] STACK 0x2003FB10:    0x1003B235 0x00000000 0x2003FB44 0x00000000
[3629373ms] STACK 0x2003FB20:    0x20011918 0x200359D0 0xD0000128 0x00000000
[3629374ms] STACK 0x2003FB30:    0x00000000 0x1002805D 0x200133B8 0x200359D0
[3629374ms] STACK 0x2003FB40:    0x00000930 0x200133B8 0x000052B0 0x00000930
[3629375ms] STACK 0x2003FB50:    0x00000001 0xFFFFFFF9 0x200133B8 0x2001E738
[3629376ms] STACK 0x2003FB60:    0x00000930 0x00000001 0x10026BD1 0x20010519
[3629377ms] STACK 0x2003FB70:    0x10026BD2 0x21000000 0x200133B8 0x20010519
[3629377ms] STACK 0x2003FB80:    0x2003FBAC 0x1002703F 0x2003FBAC 0x2002F814
[3629378ms] STACK 0x2003FB90:    0x2003FBB4 0x1002703F 0x2003FBB4 0x2002F814
[3629379ms] STACK 0x2003FBA0:    0x00000000 0x0000E09F 0x20418803 0x0001FC00
[3629380ms] STACK 0x2003FBB0:    0x20020001 0x50200000 0x00000001 0x0000E080
[3629381ms] STACK 0x2003FBC0:    0x20418803 0x0001FC00 0x00000001 0x50200000
[3629381ms] STACK 0x2003FBD0:    0x00000000 0x00000000 0x2002F7B0 0x10026BD7
[3629382ms] STACK 0x2003FBE0:    0x2002F7B0 0x20010519 0x00000100 0x20000FAF
[3629383ms] STACK 0x2003FBF0:    0x00000000 0x2003FC2C 0x20011974 0x00000001
[3629384ms] -- Command handler failed for IDE_CMD_PACKET`

aperezbios, 2025-04-02:
@PetteriAimonen as time allows, please take a look at this backtrace and see if it sheds any light on what's causing it to get wedged.

PetteriAimonen, 2025-04-07:
It seems to be stuck in some command handler, after which the watchdog handler triggers and the firmware itself recovers. But apparently the host is still stuck.

Is it possible to have the full debug log file, from beginning to the crash? I can also try to reproduce the issue here, if the image file is available somewhere?

timesscar, 2025-04-07:
I will get a clean attempt so that I can attach the full log.  If you want to reproduce locally, I tried this image; https://archive.org/details/Fallout2USA

timesscar, 2025-04-07:
[zululog.txt](https://github.com/user-attachments/files/19638166/zululog.txt)

sashahart, 2025-04-09:
I attempted to reproduce on my Pentium 3 (Abit VH6T) running Windows 98 SE, and Fallout 2 installed perfectly here (quite fast, too). I am certainly not saying OP is not experiencing a real issue - just trying to help "bisect" the problem a bit more. I'd certainly hope the ZuluIDE could work perfectly with the ITX Llama.

PetteriAimonen, 2025-04-10:
@timesscar I also tried Fallout 2 install on Windows 98 SE (Pentium 3 here too), it worked without problems.

For some reason the logs you have provided so far do not have debug logging enabled. You can see this in the beginning: `[11ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0`. If you can double-check your DIP switch settings to make sure the DEBUG switch is on, the log would be more useful. But it may be that this problem is too obscure to resolve purely based on logs, and we will need a way to reproduce it.

Because ITX-Llama does not have built-in IDE port, can you tell what brand and model of IDE expansion card or IDE-SATA adapter are you using?

timesscar, 2025-04-10:
I followed the instructions I was given to enable debug via the config file, but I will do it with the dip switches this evening.  I am using this IDE to SATA adapter https://a.co/d/2gIR0Kc

I'd chalk it up to any number of random events, but it is 100% predictable with this configuration in this case.  Will update later today with the requested logs.

PetteriAimonen, 2025-04-11:
@timesscar Thanks, I have ordered one of those for testing. It will take a week to arrive.

timesscar, 2025-04-11:
[zululog.txt](https://github.com/user-attachments/files/19707977/zululog.txt)

Log file is attached.

timesscar, 2025-04-11:
Thank you for looking into it.  It does seem like a weird edge case with the Llama and the ZuluIDE (or maybe the SATA adapter as you mention) but its just a gut feeling.

PetteriAimonen, 2025-04-24:
Seems like something is wrong with the SATA-IDE converter I got, it's not recognizing anything (not even a real IDE drive), I'll see if I can get it exchanged.

I did the largest install.
You could try some software-based CD-ROM emulator to see if that makes a difference.

PetteriAimonen, 2025-05-14:
If you have time, trying this latest release is worth a try.
It includes a few fixes that may be relevant for this problem.
https://github.com/ZuluIDE/ZuluIDE-firmware/releases/tag/v2025.05.08.1

I still haven't received the replacement IDE-SATA adapter :/

PetteriAimonen, 2025-05-21:
Thanks for testing anyway.

timesscar, 2025-05-21:
For the SATA/IDE adapter, very few work correctly with optical drives.  The one I linked is the only one I have found that does.

timesscar, 2025-12-04:
Same result in the same place.  When the install freezes, the activity light on the zuluide goes solid for several sections then turns off.

aperezbios, 2025-12-04:
@timesscar is there anything interesting in the the zululog.txt file after this happens? Could you try again and send us the resulting log file, if it's already been overwritten?

timesscar, 2025-12-05:
To clarify, was there new logging added to the latest version?