ZuluIDE/ZuluIDE-firmware #201: Intermittently working in Windows 98 SE with Triton PIIX

nizmow, 2025-05-02:
Hi there, I've had an issue with the ZuluIDE intermittently working with Windows 98 SE and Triton PIIX, both on a 430VX (Asus VX97) and now a 430HX (Gigabyte GA-586HX). BIOS detects fine, boot works, DOS works fine, and _sometimes_ things will be fine in Windows 98 SE. However,  sometimes the IDE controller the Zulu is on (secondary typically) will show yellow exclamation mark and "not working properly", and no CD-ROM will appear. At first I thought this was a board issue, but it's on two boards, and then a driver issue, but I see this WITH and WITHOUT PIIX / chipset drivers installed.

ZuluIDE log shows this:

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.03.25-release Mar 25 2025 19:48:10
[11ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[12ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x214E324330C863E6
[99ms] FPGA initialization succeeded
[99ms] I2S audio to expansion header enabled
[100ms] -- System clock is set to 135428571Hz
[101ms] Hardware UI not found.
[101ms] I2C server found
[116ms] SD card detected, FAT32 volume size: 30420 MB
[116ms] SD MID: 0x03, OID: 0x53 0x44
[117ms] SD Name: SD32G
[117ms] SD Date: 7/2024
[117ms] SD Serial: 0x8387D113
[133ms] Selecting device type when loading first image.
[144ms] Device is a CDROM drive
[144ms] Initialized platform with device control.
[146ms] Set SSID from INI file to Summerwood Cottage
[148ms] Set PASSWORD from INI file.
[148ms] Initialized platform controller with the status controller.
[161ms] Initialized platform controller with the display controller.
[161ms] Initialized platform controller with input receiver.
[173ms] Device 0 configuration:
[173ms] -- Max PIO mode: 3 (phy max 3)
[174ms] -- Max UDMA mode: 0 (phy max 0)
[174ms] -- Max blocksize: 4096 (phy max 4096)
[1194ms] Loading last used image: "win98qi_v0.9.6_ALL.iso"
[1195ms] Loading image "win98qi_v0.9.6_ALL.iso"
[1236ms] Image file win98qi_v0.9.6_ALL.iso is not contiguous, access will be slower
[1243ms] Initialization complete!
[1267ms] I2C Client subscribed to updates.
[2000ms] FPGA license request code: 0xE6 0x63 0xC8 0x30 0x43 0x32 0x4E 0x21 0xFF 0xFB 0xFF 0xFF 0xB6 0xE8 0x33 0xD6 0x51 
[2000ms] FPGA license accepted with status 0x80
[16367ms] Client IP address ient IP address is: 192.168.1.133
[61666ms] IDEATAPIDevice::cmd_packet() command read timeout
[61666ms] -- Command handler failed for IDE_CMD_PACKET
[71670ms] IDEATAPIDevice::cmd_packet() command read timeout
[71670ms] -- Command handler failed for IDE_CMD_PACKET
[81674ms] IDEATAPIDevice::cmd_packet() command read timeout
[81674ms] -- Command handler failed for IDE_CMD_PACKET
[237203ms] IDEATAPIDevice::cmd_packet() command read timeout
[237203ms] -- Command handler failed for IDE_CMD_PACKET
[247207ms] IDEATAPIDevice::cmd_packet() command read timeout
[247207ms] -- Command handler failed for IDE_CMD_PACKET
[257211ms] IDEATAPIDevice::cmd_packet() command read timeout
[257211ms] -- Command handler failed for IDE_CMD_PACKET
```

Not sure useful. The second batch I assume was when I opened explorer or maybe device manager to poke around.

Any ideas?

aperezbios, 2025-05-02:
@nizmow thanks for the report. I'd like to have you enable debug mode, which can either be done via the DIP switch or by setting debug=1 in zuluide.ini, and then attempt to replicate the error with your hardware. 

However, before you do that, please update to the latest firmware, which you can download from https://github.com/ZuluIDE/ZuluIDE-firmware/releases/download/v2025.04.18/ZuluIDE_RP2040_2025-04-18.bin

Just place the .bin file on the SD card and power on your ZuluIDE as normal. It will detect the firmware and self-update, then delete the .bin and re-boot.

nizmow, 2025-05-02:
I just upgraded after posting this actually! I’ll do some more debugging over the next few days, out of town for the moment. Thanks for the follow up.

nizmow, 2025-05-07:
These logs looked similar to this issue: https://github.com/ZuluIDE/ZuluIDE-firmware/issues/202, however I tried the attached firmware and it didn't seem to help.

aperezbios, 2025-05-08:
@nizmow can you send us a debug log but with the latest devlopment build of the firmware flashed, which you can grab from https://github.com/ZuluIDE/ZuluIDE-firmware/releases/download/latest/ZuluIDE_RP2040_2025-05-06_6d0427e.bin

Thanks!

nizmow, 2025-05-10:
Hey, that's a 404 but I realised the current release build seems to be newer so I used that, I don't seem to get errors anymore although it still doesn't work properly (same symptoms, no drive detected in Windows and a controller error in device manager). I tried a real CD-ROM and it works fine. I may try on a different Pentium motherboard when I get a chance.

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.05.08-release May  8 2025 21:23:37
[11ms] DIP switch settings: cablesel 0, drive_id 1 debug log 1
[12ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x214E324330C863E6
[13ms] DBG Loading FPGA bitstream
[99ms] FPGA initialization succeeded
[100ms] I2S audio to expansion header enabled
[101ms] -- System clock is set to 135428571Hz
[102ms] Hardware UI not found.
[102ms] I2C server found
[116ms] SD card detected, FAT32 volume size: 30420 MB
[116ms] SD MID: 0x03, OID: 0x53 0x44
[117ms] SD Name: SD32G
[117ms] SD Date: 7/2024
[117ms] SD Serial: 0x8387D113
[132ms] Selecting device type when loading first image.
[144ms] Device is a CDROM drive
[152ms] Device 1 configuration:
[153ms] -- Max PIO mode: 3 (phy max 3)
[153ms] -- Max UDMA mode: 0 (phy max 0)
[153ms] -- Max blocksize: 4096 (phy max 4096)
[168ms] DBG -- Operating as secondary drive
[168ms] Initialized platform with device control.
[170ms] Set SSID from INI file to Summerwood Cottage
[171ms] Set PASSWORD from INI file.
[171ms] Initialized platform controller with the status controller.
[173ms] Initialized platform controller with the display controller.
[173ms] Initialized platform controller with input receiver.
[185ms] Device 1 configuration:
[185ms] -- Max PIO mode: 3 (phy max 3)
[186ms] -- Max UDMA mode: 0 (phy max 0)
[186ms] -- Max blocksize: 4096 (phy max 4096)
[201ms] DBG -- Operating as secondary drive
[1205ms] Loading last used image: "win98qi_v0.9.6_ALL.iso"
[1206ms] Loading image "win98qi_v0.9.6_ALL.iso"
[1206ms] DBG Image file win98qi_v0.9.6_ALL.iso size 1273720832
[1247ms] Image file win98qi_v0.9.6_ALL.iso is not contiguous, access will be slower
[1253ms] Initialization complete!
[1260ms] DBG IDE_EVENT_HWRST
[1271ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0xFF ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[1278ms] I2C Client subscribed to updates.
[1370ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[2000ms] FPGA license request code: 0xE6 0x63 0xC8 0x30 0x43 0x32 0x4E 0x21 0xFF 0xDF 0xFF 0xDF 0xAC 0x22 0x3F 0x2D 0xDA 
[2000ms] FPGA license accepted with status 0x80
[4256ms] Loading image "win98qi_v0.9.6_ALL.iso"
[4256ms] DBG Image file win98qi_v0.9.6_ALL.iso size 1273720832
[4297ms] Image file win98qi_v0.9.6_ALL.iso is not contiguous, access will be slower
[4301ms] DBG Image file win98qi_v0.9.6_ALL.iso size 1273720832
[4341ms] Image file win98qi_v0.9.6_ALL.iso is not contiguous, access will be slower
[4342ms] -- Device loading media: "win98qi_v0.9.6_ALL.iso"
[5054ms] Client IP address is: 192.168.1.133
[9143ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0x00 DEVCTRL:0x00 ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[9154ms] DBG -- IDE regs: STATUS:0x00 CMD:0xFF DEV:0xA0 DEVCTRL:0x00 ERROR:0x01 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[10412ms] DBG IDE Command for DEV1: 0xA1 IDE_CMD_IDENTIFY_PACKET_DEVICE (device 0xB0, dev_ctrl 0x00, feature 0xAA, sector_count 0x0A, lba 0xEB 0x14 0x01)
[10811ms] DBG -- Command complete
[10822ms] DBG -- IDE regs: STATUS:0x40 CMD:0xA1 DEV:0xB0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0xAA LBAL:0x01 LBAM:0x14 LBAH:0xEB
[17731ms] DBG -- IDE regs: STATUS:0x40 CMD:0xA1 DEV:0xB0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0xAA LBAL:0x01 LBAM:0xAA LBAH:0xEB
[31595ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x08 0x00 0x01)
[31596ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x00 0x11 0x00 0x00 0x01 0x00 0x00 0x00 
[31597ms] DBG -- Reporting UNIT_ATTENTION condition after reset/medium change (ASC:0x3A00)
[31598ms] DBG -- Command complete
[31602ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x08 0x00 0x01)
[31603ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x00 0x11 0x00 0x00 0x01 0x00 0x00 0x00 
[31604ms] DBG -- Read 1 sectors starting at 17
[31605ms] DBG ---- Read CD: 1 sectors starting at 17, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 34816
[31608ms] DBG -- ATAPI success
[31608ms] DBG -- Command IDE_CMD_PACKET completed with error status 0x64
[31612ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x08 0x00 0x01)
[31613ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x00 0x8E 0x00 0x00 0x01 0x00 0x00 0x00 
[31614ms] DBG -- Read 1 sectors starting at 142
[31615ms] DBG ---- Read CD: 1 sectors starting at 142, track number 1, sector size 2048, main channel 0x10, sub channel 0x00, data offset in file 290816
[31618ms] DBG -- ATAPI success
[31618ms] DBG -- Command complete
[31625ms] DBG IDE Command for DEV1: 0xA0 IDE_CMD_PACKET (device 0xB0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0x08 0x00 0x01)
[31626ms] DBG -- ATAPI command: ATAPI_CMD_READ10 0x28 0x00 0x00 0x00 0x00 0x8F 0x00 0x00 0x01 0x00 0x00 0x00 
[31627ms] DBG -- Read 1 sectors starting at 143
[... cut ...]
[31656ms] DBG -- IDE regs: STATUS:0x50 CMD:0xA0 DEV:0xB0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0x00 LBAL:0x01 LBAM:0xFE LBAH:0xFF
[50994ms] DBG -- IDE regs: STATUS:0x50 CMD:0xEC DEV:0xA0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0x00 LBAL:0x01 LBAM:0xAA LBAH:0xFF
[52083ms] DBG -- IDE regs: STATUS:0x50 CMD:0x20 DEV:0xA0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0x00 LBAL:0x01 LBAM:0x00 LBAH:0x00
[59486ms] DBG -- IDE regs: STATUS:0x50 CMD:0xEC DEV:0xA0 DEVCTRL:0x00 ERROR:0x00 FEATURE:0x00 LBAL:0x01 LBAM:0x00 LBAH:0x00
```

aperezbios, 2025-05-10:
@nizmow thanks. Unfortunately there's nothing in the log to suggest there's any sort of problem, and we've yet to be able to replicate the issue under Windows 98SE, which is, frankly, odd. We'll have to do some more testing. It would be helpful to know if Device Manager has any further details about what exactly the problem is, if possible.

nizmow, 2025-05-12:
I tried this again, I actually removed the controller, rebooted and let Windows reinstall it. It worked, until next reboot where it failed again in the same way (the info Windows gives is just "this device is not working properly", not handy). However, I went digging and found [this old thread](https://github.com/ZuluIDE/ZuluIDE-firmware/issues/68) where an old firmware is suggested to fix an issue. Gave me the idea to try and old firmware, so I installed it and boom, it's rock solid no issues. I'm happy with this and see no real reason to upgrade firmware, so I'll close this out!