ZuluIDE/ZuluIDE-firmware #203: Sudden failure to play CD Audio.

dzlia, 2025-05-04:
While playing a game with CD Audio (Lamentation Sword), music dropped into looping few samples. All the time zulu's LED was constantly green (no blinking). ~10 seconds after music was released, but LED just got dark (no blinking). The next track has not started. Here is what I see in zululog.txt, no zuluerr.txt. DMA mode is turned on, the game doesn't seem to read anything from the disc in parallel. Since my current computer is stable, I'm reporting this just in case it could be a bug in zulu firmware.

```
[2243ms] Initialization complete!
[2247ms] FPGA license request code: 0xE6 0x63 0xC8 0x30 0x43 0x43 0x83 0x21 0xFF 0xFF 0xFF 0xFF 0x67 0xD3 0xF2 0x02 0xB9 
[2248ms] FPGA license accepted with status 0x83
[20251ms] -- Set PIO transfer mode 3
[20255ms] -- Set UDMA transfer mode 0
[5097865ms] IDEATAPIDevice::cmd_packet() command read timeout
[5097865ms] -- Command handler failed for IDE_CMD_PACKET
```

morio, 2025-05-12:
@dzlia could you try again with debug turned on. Either by the dip switch on the board or by setting `debug = 1` under the `[IDE]` section in the `zuluide.ini` file. Thank you.

dzlia, 2025-05-13:
Ok will try. Just as a small update - it has happened again few days later after a couple of hours of gameplay in the same game.

dzlia, 2025-05-25:
Just another update: I don't face this problem after enabling debug logging. This could be an indicator of a data race that causes problems when debug logging is disabled, which is "cured" by a delay on writing data in the debug mode.

dzlia, 2025-05-29:
While there was nothing that weird as written in the initial description (but I faced once case where music just stopped playing without itching), attaching debug log in which I found quite a number of non-zero error codes. Maybe it could be of some use.

[zululog.txt.gz](https://github.com/user-attachments/files/20510939/zululog.txt.gz)

morio, 2025-06-17:
@dzlia 
Had a chance to dive into the debug log. 
Was this debug log taken when it stopped playing CD audio in the middle of the game? 
Or have you still never had the audio stop when debug is on and this log was just one that had errors?

If it just contained errors, I think concerning point of interest is the CRC error in this point of the log. The system seems to recover fine though. 
```
[378658ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x01, sector_count 0x01, lba 0xFF 0xFF 0x01)
[378659ms] DBG -- ATAPI command: ATAPI_CMD_READ_SUB_CHANNEL 0x42 0x02 0x40 0x01 0x00 0x00 0x00 0x00 0x10 0x00 0x00 0x00 
[378660ms] DBG ------ Get audio playback position: status 17 lba 195858
[378661ms] DBG ------ ABS M 0x2B S 0x21 F 0x21
[378661ms] DBG ------ REL M 0x00 S 0x1D F 0x49
[378662ms] DBG ---- ATAPI send 1x16 bytes: 0x00 0x11 0x00 0x0C 0x01 0x10 0x0D 0x01 0x00 0x2B 0x21 0x21 0x00 0x00 0x1D 0x49 
[378663ms] DBG -- ATAPI success
[378663ms] DBG -- Command complete
[378664ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[378665ms] DBG -- ATAPI command: ATAPI_CMD_SEEK10 0x2B 0x00 0x00 0x02 0xFD 0x12 0x00 0x00 0x00 0x00 0x00 0x00 
[378666ms] DBG ------ CD-ROM Play Audio request at 195858 for 0 sectors
[378753ms] DBG ------ Play audio CD: 0 sectors starting at 195858, track number 13, data offset in file 460658016
[378755ms] DBG -- ATAPI success
[378755ms] DBG -- ATAPI success
[378755ms] DBG -- Command complete
[378755ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x01, sector_count 0x01, lba 0xFF 0xFE 0x01)
[378757ms] DBG -- ATAPI command: ATAPI_CMD_READ_SUB_CHANNEL 0x42 0x02 0x40 0x01 0x00 0x00 0x00 0x00 0x10 0x00 0x00 0x00 
[378758ms] DBG ------ Get audio playback position: status 21 lba 195858
[378759ms] DBG ------ ABS M 0x2B S 0x21 F 0x21
[378759ms] DBG ------ REL M 0x00 S 0x1D F 0x49
[378760ms] DBG ---- ATAPI send 1x16 bytes: 0x00 0x15 0x00 0x0C 0x01 0x10 0x0D 0x01 0x00 0x2B 0x21 0x21 0x00 0x00 0x1D 0x49 
[378761ms] WARNING: UltraDMA CRC mismatch, calculated 0xD6D9, host sent 0x8E2F
[378762ms] -- Detected 1 CRC errors during transfer, reporting error to host
[378762ms] DBG -- ATAPI error: 0x04 0x0803 (HARDWARE_ERROR)
[378763ms] DBG -- Command complete
[378763ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x01, sector_count 0x01, lba 0xFF 0xFF 0x01)
[378765ms] DBG -- ATAPI command: ATAPI_CMD_REQUEST_SENSE 0x03 0x00 0x00 0x00 0x0E 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[378766ms] DBG ---- ATAPI send 1x14 bytes: 0xF0 0x00 0x04 0x00 0x00 0x00 0x00 0x0B 0x00 0x00 0x00 0x00 0x08 0x03 
[378767ms] DBG -- ATAPI success
[378768ms] DBG -- Command IDE_CMD_PACKET completed with error status 0x44
[552788ms] DBG IDE Command for DEV0: 0xA0 IDE_CMD_PACKET (device 0xA0, dev_ctrl 0x00, feature 0x00, sector_count 0x01, lba 0xFF 0xFF 0x01)
[552789ms] DBG -- ATAPI command: ATAPI_CMD_TEST_UNIT_READY 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
[552791ms] DBG -- ATAPI success
[552791ms] DBG -- Command complete
```

The CRC glitch and the first post's error
`[5097865ms] IDEATAPIDevice::cmd_packet() command read timeout
[5097865ms] -- Command handler failed for IDE_CMD_PACKET`
 might be a timing issue due to the faster clock for audio processing. 

Try this firmware:
[zuluide-udma-div-9.zip](https://github.com/user-attachments/files/20766000/zuluide-udma-div-9.zip)

and see if you get any errors or audio playback issues.

dzlia, 2025-06-17:
@morio , in the debug mode I've never faced exactly same effect as in the original post. However, I noticed once that music stopped playing while I was away. I'm not sure if it was the same problem or not, because there was no stuttering and I noticed this after music stopped playing already, because I was mentally in the game.
I have finished with the game that caused problems but I'll try to install it again to test your patch.

dzlia, 2025-06-18:
Here is the video with CD player.

https://github.com/user-attachments/assets/7caae694-1ed4-4769-b4dd-e3f5d0198834