ZuluIDE/ZuluIDE-firmware #211: Max_volume setting has no effect

Aardvark1982, 2025-05-12:
I have tried "max_volume = 100" and "max_volume = 10", and both are equally loud.
I don't see any reference to this setting in Zululog.

[zululog.txt](https://github.com/user-attachments/files/20168401/zululog.txt)

[ZuluIDE.ini.txt](https://github.com/user-attachments/files/20168402/ZuluIDE.ini.txt)

v2025.05.08 firmware

morio, 2025-05-12:
Thanks for the catch.
Try this fix: https://github.com/ZuluIDE/ZuluIDE-firmware/actions/runs/14981082344
It should print out the max volume setting in the log now. It will be printed after "Initialization complete!" to guarantee the setting is read after the SD card is inserted for the first time.

Aardvark1982, 2025-05-12:
Thanks, working fine now.

closed 2025-05-19 by morio