ZuluIDE/ZuluIDE-firmware #217: Large (>4GB?) iso not working correctly

ric96, 2025-06-30:
I am trying to boot some large ISOs on a G41 LGA775 based platform.

I have tried the following Images:
1. debian-edu-12.11.0-amd64-BD-1.iso 7.6GB
2. [debian-edu-12.11.0-i386-BD-1.iso](https://cdimage.debian.org/debian-cd/12.11.0/i386/iso-bd/debian-edu-12.11.0-i386-BD-1.iso) 7.7GB
3. CentOS-8.2.2004-x86_64-dvd1.iso 7.7GB

The ISOs "boot" just fine but when loading the installer all fail.
The Centos image fails at  the image validation part at around 52%.
Both debian images fail at failing to copy installer components form installation media after the GUI has loaded up.

I am using the SDCard formatted as exFAT so the images copy over fine and zuluide ofc detects and loads them fine.

Here's the log:
```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.06.19-release Jun 19 2025 23:41:27
[11ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[12ms] Flash chip size: 2048 kB
[13ms] Flash unique ID: 0x375E6993476464E6
[99ms] FPGA initialization succeeded
[99ms] I2S audio to expansion header enabled
[100ms] -- System clock is set to 135428571Hz
[101ms] Hardware UI not found.
[101ms] I2C server not found
[180ms] SD card detected, FAT64 volume size: 121936 MB
[180ms] SD MID: 0x03, OID: 0x53 0x44
[180ms] SD Name: SD128
[180ms] SD Date: 3/2025
[181ms] SD Serial: 0xE2778356
[214ms] Device is a CDROM drive
[252ms] Device 0 configuration:
[252ms] -- Max PIO mode: 3 (phy max 3)
[253ms] -- Max UDMA mode: 0 (phy max 0)
[253ms] -- Max blocksize: 4096 (phy max 4096)
[362ms] Device 0 configuration:
[362ms] -- Max PIO mode: 3 (phy max 3)
[362ms] -- Max UDMA mode: 0 (phy max 0)
[362ms] -- Max blocksize: 4096 (phy max 4096)
[453ms] Loading last used image: "debian-edu-12.11.0-i386-BD-1.iso"
[454ms] Loading image "debian-edu-12.11.0-i386-BD-1.iso"
[456ms] Image file debian-edu-12.11.0-i386-BD-1.iso is not contiguous, access will be slower
[473ms] Initialization complete!
[481ms] Max volume set to 100/100
[2000ms] FPGA license request code: 0xE6 0x64 0x64 0x47 0x93 0x69 0x5E 0x37 0xFF 0xFF 0xFF 0xFF 0x47 0x22 0x19 0xCA 0xDC 
[2000ms] FPGA license accepted with status 0x81
[3475ms] Loading image "debian-edu-12.11.0-i386-BD-1.iso"
[3482ms] Image file debian-edu-12.11.0-i386-BD-1.iso is not contiguous, access will be slower
[3488ms] Image file debian-edu-12.11.0-i386-BD-1.iso is not contiguous, access will be slower
[3488ms] -- Device loading media: "debian-edu-12.11.0-i386-BD-1.iso"
[17709ms] -- Set UDMA transfer mode 0
[74726ms] -- Set UDMA transfer mode 0
```

ric96, 2025-06-30:
I did some tests that confirm my 4GB limit issue. So there seems to be some 32bittery going on or something imposing the single layer 4.7GB limit?

I created two iso.
1. Has 2x 2GB test images. [Link to test iso](https://people.linaro.org/~sahaj.sarup/iso2Gtest.iso)
2. Has 4x 2GB test images  [Link to test iso](https://people.linaro.org/~sahaj.sarup/iso2Gtest2.iso)

All test images are created from /dev/urandom so they are not blank.

The first ISO, that is 4GB in size and has 2x 2GB test images passes checksum fine:

[image]

The second ISO that is 8GB in size and has 4x 2GB images, passes for only the first 2 files and then fails for the rest.

[image]

FWIW these tests were done with the zuluide attached to a 64bit debian system and the logs remain the same.

aperezbios, 2025-06-30:
@ric96 thanks for the report. We were able to reproduce the issue, and this should now be fixed in this build. If you could independently confirm this, it would be greatly appreciated. I'm closing this issue for now, but feel free to re-open it if you observe the same or similar issues.

Please download the .zip file, extract the .bin file to your ZuluIDE SD card, then power on the ZuluIDE, and it will initiate a self-update.

[ZuluIDE_RP2040_2025-06-30_c292d4b.zip](https://github.com/user-attachments/files/20987931/ZuluIDE_RP2040_2025-06-30_c292d4b.zip)

closed 2025-06-30 by aperezbios

InterClaw, 2025-07-01:
> [@aperezbios](https://github.com/aperezbios) thanks! perfect, this fixes the issue for me. Tested on all the iso I listed above. I wonder if it fixes [#156](https://github.com/ZuluIDE/ZuluIDE-firmware/discussions/156) as well.

No such luck I'm afraid. Tried it out now. But it was a long shot anyways. :)

aperezbios, 2025-07-01:
@InterClaw thanks for letting us know. I'll ask some more questions now in the issue you filed :)