ZuluIDE/ZuluIDE-firmware #219: Haiku iso Boot issue

ric96, 2025-07-01:
Having issues getting [haiku](https://www.haiku-os.org/get-haiku/r1beta5/) to boot.

Initial boot is fine but then it crashes.

[image]

I get the following errors in the zuluide log

```
[11ms] Platform: ZuluIDE RP2040
[11ms] FW Version: 2025.07.01-dev Jun 30 2025 22:27:47
[11ms] DIP switch settings: cablesel 0, drive_id 0 debug log 0
[12ms] Flash chip size: 2048 kB
[12ms] Flash unique ID: 0x375E6993476464E6
[99ms] FPGA initialization succeeded
[99ms] I2S audio to expansion header enabled
[100ms] -- System clock is set to 135428571Hz
[101ms] Hardware UI not found.
[101ms] I2C server not found
[208ms] SD card detected, FAT64 volume size: 121936 MB
[209ms] SD MID: 0x03, OID: 0x53 0x44
[209ms] SD Name: SD128
[209ms] SD Date: 3/2025
[209ms] SD Serial: 0xE2778356
[243ms] Device is a CDROM drive
[281ms] Device 0 configuration:
[281ms] -- Max PIO mode: 3 (phy max 3)
[282ms] -- Max UDMA mode: 0 (phy max 0)
[282ms] -- Max blocksize: 4096 (phy max 4096)
[390ms] Device 0 configuration:
[390ms] -- Max PIO mode: 3 (phy max 3)
[391ms] -- Max UDMA mode: 0 (phy max 0)
[391ms] -- Max blocksize: 4096 (phy max 4096)
[483ms] Loading last used image: "haiku-r1beta5-x86_64-anyboot.iso"
[483ms] Loading image "haiku-r1beta5-x86_64-anyboot.iso"
[561ms] Initialization complete!
[570ms] Max volume set to 100/100
[2000ms] FPGA license request code: 0xE6 0x64 0x64 0x47 0x93 0x69 0x5E 0x37 0xFF 0xFF 0xFF 0xFF 0x47 0x22 0x19 0xCA 0xDC 
[2000ms] FPGA license accepted with status 0x81
[3563ms] Loading image "haiku-r1beta5-x86_64-anyboot.iso"
[3578ms] -- Device loading media: "haiku-r1beta5-x86_64-anyboot.iso"
[17905ms] -- Set UDMA transfer mode 0
[41563ms] --------------
[41563ms] WATCHDOG TIMEOUT, attempting bus reset
[41563ms] GPIO states: out 0x00FE0803 oe 0x0207007F in 0x3DFB0AFB
[41564ms] STACK 0x2003FB28:    0x1003B6D6 0x3DFB0AFB 0x20012120 0x2003FB6C
[41565ms] STACK 0x2003FB38:    0x1003B6D6 0x00000000 0x20012134 0x00000000
[41565ms] STACK 0x2003FB48:    0x20012120 0x200364D7 0xD0000128 0x00000000
[41566ms] STACK 0x2003FB58:    0x00000000 0x10028345 0x50200000 0x200364D7
[41567ms] STACK 0x2003FB68:    0x00000002 0x50200000 0x00000001 0x00000002
[41568ms] STACK 0x2003FB78:    0xD0000000 0xFFFFFFF9 0x50200000 0x00000000
[41568ms] STACK 0x2003FB88:    0x00000002 0x00000001 0x10029131 0x20000CE7
[41569ms] STACK 0x2003FB98:    0x10029130 0x01000000 0x00000001 0x0001FC00
[41570ms] STACK 0x2003FBA8:    0x00000001 0x00000000 0x00000000 0x00000000
[41571ms] STACK 0x2003FBB8:    0x2003FC0F 0x2001C580 0x00000000 0x20000F55
[41571ms] STACK 0x2003FBC8:    0x00000100 0x20000FAF 0x00000000 0x00000001
[41572ms] STACK 0x2003FBD8:    0x2003FC0F 0x00000000 0x2003FC17 0x00000000
[41573ms] STACK 0x2003FBE8:    0x00000000 0x00000000 0x2003FC0F 0x2001C580
[41574ms] STACK 0x2003FBF8:    0x0000001B 0x20001895 0x00000000 0x20001DAD
[41574ms] STACK 0x2003FC08:    0x00000000 0x2001C580 0x00000800 0x00000800
[41575ms] STACK 0x2003FC18:    0x00000000 0x20008101 0x20013B98 0x0000001B
[41620ms] -- Command handler failed for IDE_CMD_PACKET
[56576ms] --------------
[56576ms] WATCHDOG TIMEOUT, attempting bus reset
[56576ms] GPIO states: out 0x00FE0803 oe 0x0207007F in 0x3FFB0AFB
[56577ms] STACK 0x2003FB08:    0x1003B6D6 0x3FFB0AFB 0x2003FB3C 0x10027327
[56578ms] STACK 0x2003FB18:    0x1003B6D6 0x00000000 0x2003FB44 0x00000000
[56579ms] STACK 0x2003FB28:    0x2003FB44 0x200364D7 0xD0000128 0x00000000
[56579ms] STACK 0x2003FB38:    0x00000000 0x10028345 0x502000CC 0x200364D7
[56580ms] STACK 0x2003FB48:    0x502000DC 0x00000000 0x50200000 0x000000DC
[56581ms] STACK 0x2003FB58:    0x00000060 0xFFFFFFF9 0x502000CC 0x0000E09F
[56582ms] STACK 0x2003FB68:    0x502000DC 0x20418803 0x502000DC 0x20000CF5
[56582ms] STACK 0x2003FB78:    0x10029198 0x01000000 0x20418803 0x0000E09F
[56583ms] STACK 0x2003FB88:    0x20418803 0x0001FC00 0x00000001 0x50200000
[56584ms] STACK 0x2003FB98:    0x00000001 0x00000002 0xD0000000 0x20000CF5
[56585ms] STACK 0x2003FBA8:    0x00000001 0x50200000 0x00000001 0x00000000
[56586ms] STACK 0x2003FBB8:    0x20418803 0x2003FC17 0x00000000 0x2001C580
[56586ms] STACK 0x2003FBC8:    0x00000000 0x20000F55 0x000000FF 0x20000C9D
[56587ms] STACK 0x2003FBD8:    0x00000000 0x00000001 0x2003FC17 0x00000000
[56588ms] STACK 0x2003FBE8:    0x00000000 0x00000000 0x2003FC0F 0x2003FC17
[56589ms] STACK 0x2003FBF8:    0x00000000 0x2001C580 0x0000001B 0x20001DAD
[71589ms] --------------
[71589ms] WATCHDOG TIMEOUT, attempting bus reset
[71590ms] GPIO states: out 0x00FE0801 oe 0x0207007F in 0x3DFB0A89
[71591ms] STACK 0x2003FB58:    0x1003B6D6 0x3DFB0A89 0x20012134 0x1002AED7
[71591ms] STACK 0x2003FB68:    0x1003B6D6 0x00000000 0x20013B98 0x00000000
[71592ms] STACK 0x2003FB78:    0x2003FBA4 0x200364D7 0xD0000128 0x00000000
[71593ms] STACK 0x2003FB88:    0x00000000 0x10028345 0x50000000 0x200364D7
[71594ms] STACK 0x2003FB98:    0x00000100 0x50200000 0x00000000 0x000000FF
[71594ms] STACK 0x2003FBA8:    0x00000000 0xFFFFFFF9 0x50000000 0x0F000E02
[71595ms] STACK 0x2003FBB8:    0x00000100 0x00000000 0x502000DC 0x20000F7B
[71596ms] STACK 0x2003FBC8:    0x20000FB2 0x61000000 0x00000000 0x20000C9D
[71597ms] STACK 0x2003FBD8:    0x00000000 0x00000001 0x2003FC17 0x00000000
[71597ms] STACK 0x2003FBE8:    0x00000000 0x00000000 0x2003FC0F 0x2003FC17
[71598ms] STACK 0x2003FBF8:    0x00000000 0x2001C580 0x0000001B 0x20001DAD
[71599ms] STACK 0x2003FC08:    0x00000000 0x2001C580 0x00000800 0x0000001B
[71600ms] STACK 0x2003FC18:    0x00000800 0x200080E7 0x20013B98 0x0000001B
[71600ms] STACK 0x2003FC28:    0x00000800 0x20013B98 0x00002800 0x00000800
[71601ms] STACK 0x2003FC38:    0x0000001B 0x200092CB 0x2003FC80 0x20030318
[71602ms] STACK 0x2003FC48:    0x200302B0 0xFFFFFFF9 0x2001C580 0x00000800

```

aperezbios, 2025-07-08:
I was able to replicate this [zululog-haiku-boot-watchdogtimeout.txt](https://github.com/user-attachments/files/21132409/zululog-haiku-boot-watchdogtimeout.txt), @PetteriAimonen can you take a look when you're able/back? Thanks.

PetteriAimonen, 2025-08-08:
Yeah, this occurs for me too. I'll take a look.

PetteriAimonen, 2025-09-08:
I assume this is resolved by the fix above.

closed 2025-09-08 by PetteriAimonen